motioneye-project/motioneyeos #147: Unable to open video device

jolabpi, 2016-02-01:
Hi,
I haven no success in getting a videostream in my browser although with VLC on Raspbian I can get a stream from my IP camera as well as from webcams(with v4l2). Using motioeyes I can add the cameras (using same URL as with VLC), the cameras are added to the viewer but the screen remains grey "Unable to open video device). I also tried out the emotioneyesOS on my raspberri PI2. Same behaviour. I'm at a loss how to proceed further. Do you have some tips ?
Thnx
Johan

Dobi29, 2016-02-04:
i have the same problem. by using http://sourceforge.net/projects/onvifdm i get the url of my ip-cam. with motioneyeos i get the rstp stream, under raspbian / motioneye the same error. 

do you use this instruction ? https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

do you know how i can check using mrdave ?

ccrisan, 2016-02-04:
Run `motion -h` and let us know the version it prints out.

Dobi29, 2016-02-04:
what are the steps to install motion / mrdave on a new raspbian installation ? i try it this evening...

-sudo apt-get install motion and then note 2: A prebuilt version of Mr Dave's motion for Raspbian can be downloaded from here. Just download it and copy it to /usr/local/bin/motion.

that's all to run mrdave ?

ccrisan, 2016-02-04:
Yes!

jolabpi, 2016-02-04:
using ssh from my android phone this is what is returned by MotionEyesOS on running the motion-h command: 

using motion binary /usr/bin/motion-mrdave

jolabpi, 2016-02-04:
for motioneye on raspbian I will redo a full clean install but already the first time  I followed your instructions to install motion-mrdave without result. Maybe I overlooked something ?

Dobi29, 2016-02-04:
a few days a made a installation following your instructions. by using motion -h i got  "motion 3.2.12+git20140228". that's why i ask for the steps to install of mrdave...

jolabpi, 2016-02-04:
Running MotionEyesOS from SSH with command :  motion

login as: admin
Welcome to meye-a187e967!
[root@meye-a187e967 ~]# motion
using motion binary /usr/bin/motion-mrdave
[0] [ALR] [ALL] conf_load: could not open configfile /etc/motion/motion.conf: No such file or directory
[0] [CRT] [ALL] conf_load: Not config file to process using default values
[0] [NTC] [ALL] motion_startup: Using default log level (NTC) (6)
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-UNKNOWN Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0] [NTC] [ENC] ffmpeg_init: ffmpeg LIBAVCODEC_BUILD 3680612 LIBAVFORMAT_BUILD 3679332
[0] [NTC] [ALL] main: Thread 1 is device: /dev/video0 input -1
[0] [NTC] [ALL] main: Stream port 0
[0] [NTC] [ALL] main: Waiting for threads to finish, pid: 1072
[1] [NTC] [ALL] motion_init: Thread 1 started , motion detection Enabled
[1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
[1] [NTC] [VID] v4l2_get_capability:
------------------------
cap.driver: "ov519"
cap.card: "USB Camera"
cap.bus_info: "usb-bcm2708_usb-1.4"
cap.capabilities=0x85200001
------------------------
[1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1] [NTC] [VID] v4l2_get_capability: - READWRITE
[1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1] [NTC] [VID] v4l2_select_input: name = "ov519", type 0x00000002, status 00000000
[1] [NTC] [VID] v4l2_select_input: - CAMERA
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [NTC] [VID] v4l2_set_pix_format: Config palette index 17 (YU12) doesn't work.
[1] [NTC] [VID] v4l2_set_pix_format: Supported palettes:
[1] [NTC] [VID] v4l2_set_pix_format: (0) O518 (O518)
[1] [NTC] [VID] v4l2_set_pix_format: 0 - O518 (compressed : 0) (0x3831354f)
[1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
[1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[1] [ERR] [ALL] motion_init: Error capturing first image
[1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1] [NTC] [VID] vid_close: Closing video device /dev/video0
[1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
[1] [NTC] [VID] v4l2_get_capability:
------------------------
cap.driver: "ov519"
cap.card: "USB Camera"
cap.bus_info: "usb-bcm2708_usb-1.4"
cap.capabilities=0x85200001
------------------------
[1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1] [NTC] [VID] v4l2_get_capability: - READWRITE
[1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1] [NTC] [VID] v4l2_select_input: name = "ov519", type 0x00000002, status 00000000
[1] [NTC] [VID] v4l2_select_input: - CAMERA
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [NTC] [VID] v4l2_set_pix_format: Config palette index 17 (YU12) doesn't work.
[1] [NTC] [VID] v4l2_set_pix_format: Supported palettes:
[1] [NTC] [VID] v4l2_set_pix_format: (0) O518 (O518)
[1] [NTC] [VID] v4l2_set_pix_format: 0 - O518 (compressed : 0) (0x3831354f)
[1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
[1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1] [NTC] [VID] vid_close: Closing video device /dev/video0
[1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
[1] [NTC] [VID] v4l2_get_capability:
------------------------
cap.driver: "ov519"
[... 20 lines cut ...]
[1] [NTC] [VID] v4l2_get_capability:
------------------------
cap.driver: "ov519"
cap.card: "USB Camera"
cap.bus_info: "usb-bcm2708_usb-1.4"
cap.capabilities=0x85200001
------------------------
[1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1] [NTC] [VID] v4l2_get_capability: - READWRITE
[1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1] [NTC] [VID] v4l2_select_input: name = "ov519", type 0x00000002, status 00000000
[1] [NTC] [VID] v4l2_select_input: - CAMERA
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [NTC] [VID] v4l2_set_pix_format: Config palette index 17 (YU12) doesn't work.
[1] [NTC] [VID] v4l2_set_pix_format: Supported palettes:
[1] [NTC] [VID] v4l2_set_pix_format: (0) O518 (O518)
[1] [NTC] [VID] v4l2_set_pix_format: 0 - O518 (compressed : 0) (0x3831354f)
[1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
[1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1] [NTC] [VID] vid_close: Closing video device /dev/video0
^C[1] [NTC] [ALL] motion_loop: End of event 1
[1] [NTC] [ALL] motion_loop: Thread exiting
[0] [NTC] [ALL] main: Threads finished
[0] [NTC] [ALL] main: Motion terminating

jolabpi, 2016-02-04:
With MotionEyesOS connecting to a networkstream rtsp:// I got video in the webbrowser. I had to remove the requirement for rtsp authentification in my camera server. Found this by looking at the motion log. It seems the url which MotioEyesOS  passes on does not contain the user:password strings as entered in the user password textboxes. Now we only have to resolve the use of using a webcam.

Dobi29, 2016-02-04:
these are my steps for running mrdave...

-sudo apt-get update
-sudo apt-get upgrade
-sudo wget https://github.com/ccrisan/motioneye/wiki/precompiled/ffmpeg_2.8.3.git325b593-1_armhf.deb
-sudo dpkg -i ffmpeg_2.8.3.git325b593-1_armhf.deb
-sudo apt-get install motion
-sudo cp /home/pi/downloads/motion-mrdave-raspbian /usr/local/bin/motion
-motion -h

[image]

where is mrdave ?

ccrisan, 2016-02-04:
You'll probably need to make it executable as well: `chmod +x /usr/local/bin/motion`.

Dobi29, 2016-02-04:
[image]

ccrisan, 2016-02-04:
That's what i'm talking about :)

Dobi29, 2016-02-04:
...that's it.. Perfect. Now it works. Thanks for your help.

ccrisan, 2016-02-05:
I'm glad it could help. I'll update the wiki instructions.

closed 2016-02-05 by ccrisan

jolabpi, 2016-02-12:
My problem with webcams was due to incompatible cams. Just tried it out now with a logitech c270, works OK. Great program.

AdrianTabacaru, 2021-07-10:
Salut, incerc sa implementez o camera sonoff gk-200mp2-b pe un sistem HomeAssistant care ruleaza pe un VirtualBox si nu reusesc sa primesc semnal video. Apare mesajul mesajul "Unable to open video device" sau o iconita cu o camera taiata.
Nelamurirea mea este, in primul rand daca MotionEye functioneaza pe Virtual Box si daca este posibil ca problema sa fie codecul ffmpeg pe care cunostintele mele de programare ma incurca in a-l instala pe HomeAssistent-ul instalat pe Oracle VM Linux (64).
Multumesc!

starbasessd, 2021-07-10:
Yes, motionEye works on VirtualBox, it's how I recreate user issues on PC hardware or 'normal linuxes.
Have you tried:
add camera
URL  rtsp://ip_address/live/ch00_0
with username and password added to the correct boxes?
There is one listing for Sonoff on iSpy (best resource for finding camera URLs IMO)
There are many known issues with HomeAssistant and motionEye.
You don't say what your base OS is for the VM, 
You could test a clean Ubuntu 21.04, and install motionEye (without HomeAssistant) using these instructions:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
If you can get it to work, then you can try adding HomeAssistant to the mix.
This issue is from 5 years ago. **Please open a new issue for further help**.