motioneye-project/motioneyeos #153: Google Drive & Date setting

DonCorleonez, 2016-02-07:
Hi Calin, this is great development.
My setup is working fine for internal access but i have some noob problem regarding google drive setting, date, & access from outside.

1. My date is incorrect (1970-1-1) but the time is correct, i am using Asia/Jakarta Time Zone

2. I have no luck trying to setting google drive for my setup, my setting is 
"Upload Media Files ON", 
"Upload Service GOOGLE DRIVE"
"Location FOLDERX" --> is this correct?
"Authorization Key XXXXXXXX"
but when i test service "**Accessing the upload service failed: !**"

3. and regarding streaming from outside network, i already open the port for http & streaming. and also i already success by trying to connect to my router from outside network (ex. mypublicip:80). But whenever i try to access MotionEye streaming (ex. mypublicip:8081), it is unavailable.

could you help me to figure this out, thanks in advance for great support

DonCorleonez, 2016-02-07:
sorry after re-format, re-write, & re-install MotionEyeOS i successfully solve point 1&2. Still need help for how to access it from outside network.

ccrisan, 2016-02-07:
Hi,

3) If I understand correctly, you have forwarded port 80 and 8081 to your PI. That's alright in theory, although I strongly recommend using something else than 80 for web access, as your PI will be the target of many unwanted requests. Nevertheless, the 8081 part should work. What do you mean it's unavailable? What error do you get? How are you trying to access it. Are you sure you need it? (it's only needed when using Fast Network Camera).

DonCorleonez, 2016-02-07:
Thx Calin, as this setup i am using as surveillance. i want to access it when i am away from home.
i'll consider to change web access port :) thx for the reminder.

i try to access from outside by input http://mypublicip:8081, and the browser came up with "The Webpage is Not Available" "err_connection_refused" i am using chrome

ccrisan, 2016-02-07:
Connection refused indicates that the port forwarding does not work for some reason. Here are some ideas:

* make sure it works inside your local network on port 8081, accessed using your private ip address (nth camera has port 808n associated)

* make sure you haven't mistyped 8081 as incoming or forwarded port in your router

* try accessing your public IP:8081 from both inside and outside of your local network and see if there are any differences

* check with your ISP to see if incoming connections on port 8081 are allowed

DonCorleonez, 2016-02-08:
Thank for your suggestion,
finally i can access by changing port to other than 8081

closed 2016-02-08 by DonCorleonez