motioneye-project/motioneyeos #156: samba & Windows 10 home issues

snagglewest, 2016-02-10:
Background:

My network is mixed OS running several different versions of MS Windows and several different flavors of Linux utilizing samba and a workgroup.  (not a fan of home group)

PVR/server = windows 7 pro
various laptops - windows Vista, 7 home, & linux mint
2 desktops linux Ubuntu 14.04 desktop / Windows 7 pro
media center - windows 7 pro
2 - OSMC/Kodi media players (RasberryPI 2)
2 - Motioneyeos (RasberryPI 2)

Over the weekend I picked up 3 low end laptops for the wife & girls.  They just check email, some light word processing and web surfing so the laptops are fairly low end with Windows 10 home installed.

While I have had some nuisance issues with various computers not showing up in the workgroup it never really presented a problem until the Windows 10 units showed up.  Once the Windows 10 units were added to the network my problems started... it seems windows 10 (at least home) does not play very nice.  This seems to be directly related to Windows 10 and is apparently a well known issue.

After a LOT of exploration I found the problem is related to the network master browser computer.  If a linux machine is elected master browser none of the other samba units show up AND windows 10 won't allow access to other windows shares.  As luck would have it my Ubuntu 14.04 desktop decided it wanted to be master browser.  So after figuring out how to fix that via the samba conf I restarted everything on the network and then one of the OSMC/Kodi boxes decided it wanted to be master... So I fixed the samba conf on both of them and rebooted everything again.  Then one of the Motioneyeos decided IT wanted to be master.  It seems samba for some reason has priority over windows... Long story short after fixing the Motioneyes and figuring out how to force the Windows 7pro PVR/server to be the master everything is finally working correctly but the linux boxes weren't showing up.  After a lot of research I found that "netbios name= XXX" needs to be added to the samba conf.  Finally success!  All members of the network show up and all are browsable. 

Anyhow, the point of my post is that the samba.conf needs to be modified to add the netbios name and a few more lines to prevent the linux from becoming the master browser.  I have it working now, but I can assume that as soon as I upgrade I'll loose all the changes and I'll have to go through the entire process again.  Would it be possible to add some advanced samba conf options?  Maybe like another tab like the "expert" settings?

------------------------
workgroup = (your workgroup name)
netbios name = (your netbios name)
preferred master = no
domain master = no
local master = no
os level = 0
------------------------

For those running windows LANscanner v1.62 (http://scottiestech.info/downloads/) is a nice utility that scans the network and finds the master browser, very handy!

Snag

ccrisan, 2016-02-10:
Nice post. I wasn't aware of these issues with Samba and Windows 10. Thanks for sharing.

I'll make sure to add these settings to the default smb.conf. Is workgroup required as well? If yes, if configured to some hardcoded value, will it work? I prefer keeping samba settings away from the Expert Settings and the netbios name can be automatically filled in with the hostname (which will soon be configurable).

snagglewest, 2016-02-10:
I'm far from a windows networking expert and even less knowledgeable about samba,  I do know now though that Windows 10 is a networking mess though and there's a ton of complaints all over the net on  dating back to the first release. As I said in the first post I use workgroup as it seems to work the best across multiple windows versions and linux.

From my experience trying to sort out the problems I found that workgroup is required as well as the netios name for a linux box to show up in the network.  You could default the netbios name could be hard coded to something like the serial number of the Pi but that is the "computer name" that shows up in the windows network browse.  I name mine "CAMERA1" & "CAMERA2" for ease of reading and it would be handy to make it configurable.

As far as the master settings I'm not absolutely positive the preferred, domain, local and OS entries are all required but I found that that combination prevented the various linuix boxes from winning the master browser election and taking over.  

Maybe a separate tab to let users define what they want?  Maybe being exposed when the user activates samba?  In my situation I want to prevent motioneyeos from becoming a master browser no mater what.  But depending on the network setup though it might be necessary for one to become a master browser.  For instance if the network was only motioneyes for a camera system.  That's my thinking, but again I only have a rudimentary understanding of networking.

Snag

ccrisan, 2016-02-20:
I made the necessary changes to smb.conf. The next version will contain these changes.

closed 2016-02-20 by ccrisan