motioneye-project/motioneyeos #157: Kodi integration

Thomas-O, 2016-02-11:
Hi,

I suspect that a great many of those who install and use Motioneyeos will also have a device running Kodi or OpenElec. [Maikito26](https://github.com/maikito26/plugin.video.surveillanceroom) has developed and released a Kodi add-on with which you can access and view an IP camera via a TV screen. It might be an idea to suggest working together so that motioneyeos will integrate easily and seamlessly with the Surveillance Room add-on.

Thanks for your time.

_Thomas_

ccrisan, 2016-02-21:
I believe the integration work should be done on the add-on side, rather than on motionEyeOS's side (or motionEye's side, for that matter). I'd be happy to help, though.

closed 2016-02-21 by ccrisan

henzo2733, 2016-05-15:
hi  anyway to get this working on kodi install?

instructions please

have motioneye working on standalone sd card swapping the sd is tedious
thanks