motioneye-project/motioneyeos #158: webserver issues

reddai75, 2016-02-11:
Hello,
first of all, great job, Motioneyeos is outstanding!
I've been experiencing this issue since the beginning: After a few hours running, I can't access anymore the configuration page (ERR_CONNECTION_REFUSED)
I've tried everything, formatting, rebooting repartitioning and reformating... 
Nothing :(

ccrisan, 2016-02-12:
Are you sure you're accessing the correct IP address?

reddai75, 2016-02-12:
Hello,

Thank you for your answer!
Yes, I always check the ip address on my router. Something get wrong but I
can't understand where. I'm still able to connect on ssh and reboot from
ssh but.... nothing else. Even after I give the reboot command it does not
work and the only access i have is ssh. No video streaming, no webserver...
:(

Marco

reddai75, 2016-02-12:
Correction: after issuing the reboot command via ssh it reboots but the problem is still there. The only option I have is to repartition reformat and reflash motioneyeos

Hexagon, 2016-02-12:
I had similar very similar problems with an older release (like mid 2015) and raspberry pi b+. Latest version works like a charm though. Which platform are you running it on, and which version?

reddai75, 2016-02-12:
Hello Hexagon thank you for your answer.
I have the issue with my raspberry pi b+ software release dates back to Jan 2016.
In a couple of hours I'll send you guys my log files. 
Thank you!

Marco

Hexagon, 2016-02-12:
Another thing to try when using B+ is to look over the power supply, especially when combining camera with an USB-wifi-stick. I've had random issues where the cable (!?) was the problem. Used a 2.1A samsung charger with a noname brand cable, replaced cable with samsung original and the problems went away :) 

Best bet is to power USB-appliances with a separate hub.

reddai75, 2016-02-12:
Thank you,
I'm actually using the original charge with 2.0 A so... It should not be
the problem I think, I have two raspberry PI and the second is running a
mediacenter flawlessy, same charger...

Marco

reddai75, 2016-02-12:
mmmhhhh how can I access the log file ? from the sd I mean...

reddai75, 2016-02-13:
Here we go, my logs

[boot.txt](https://github.com/ccrisan/motioneyeos/files/129151/boot.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/129152/motion.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/129153/motioneye.txt)

ccrisan, 2016-02-14:
I'm seeing a lot of `"/data/output" does not exist or is not writable` in your log files. I'm not really sure how that could have happened; it probably is an SD card corruption during initial boot. Please rewrite the OS image from scratch using the 2 amps PSU and make sure not to cut the power at all during setup.

reddai75, 2016-02-14:
I see your point...But I've done multiples rewrite of motioneyeos from
scratch always using diskpart first, cleaning from previous data, making a
new partition and deep formatting.  I've written the image always without
getting any issue and the raspberry runs always with the 2.0a charger...
May be I ve got a bad sd.. I'll try with a new one!

Thanks!

reddai75, 2016-02-14:
Nothing,
I used a new SD card, installed and used for 24 hours. Everything went fine until I changed the webserver port from 80 to 64000. No access anymore to the camera, no streaming, no ssh, no ftp.

ccrisan, 2016-02-14:
That's odd. Can you please attach the new logs? (please attach them to the github issue, not to the email reply).

reddai75, 2016-02-14:
I couldn't. I had no access to the camera and I've no idea how to extract the files from the sd... (any idea ?)
Anyhow, new image flashed (ssd clean and formatted)
I've configured it again and after a few reboot here is the situation:
For the moment things are working fine but I had some issues during a couple of restarts...
here are my logs:
[boot.txt](https://github.com/ccrisan/motioneyeos/files/129754/boot.txt)
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/129755/dmesg.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/129757/messages.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/129756/motion.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/129758/motioneye.txt)

nduc, 2016-02-17:
I had some issues when I first started.  But I narrowed my corruption and other issue to over clocking the RPi.  Depending on your case, your set up might be running hot and corrupts your system.  I hope it will help you.

reddai75, 2016-02-20:
Thank you,
I have the same problem also without overclocking it...
May be it is an hardware issue... I have no other explaination...

reddai75, 2016-02-20:
after so many troubleshooting, flashing, changing sd and configuration what I have now is: if I switch off the camera led when I reboot it doesn't reboot but shut down instead. With the led on, no issues up to now. I think it is an hardware issue... But I don't know hot to verify this..

ccrisan, 2016-02-20:
That sounds indeed a bit strange. Do you have the possibility to (even temporarily) replace the board?

reddai75, 2016-02-22:
Not yet,
I'm buyng a new camera with IR leds but it will take time...
i'll keep you posted, anyhow i'm 99% sure that the camera has an issue, I
remember that I put the camera on the wrong connector (display) for a few
seconds..
At first I didn't think about it but after reading few posts around I have
seen that I can damage the camera (and the board too ? )

:(

Marco

Gordon-1, 2016-02-24:
Looks like your comment about /data/output: 
(https://github.com/ccrisan/motioneyeos/issues/158#issuecomment-183851624)
is likely related to the issue I listed here (maybe wrong place):
https://github.com/ccrisan/motioneye/issues/134

But maybe thats not related to the main issue here.

closed 2017-03-01 by bortek