motioneye-project/motioneyeos #160: Connecting to a Mac SMB share

bazcurtis, 2016-02-13:
Hi,

I have been making good progress with my setup, but I just can't get it to connect to me El Capitan SMB share. I can connect to the same share from another Mac. Is this likely to be because El Capitan is using SMB v3?

Best wishes

Michael

bazcurtis, 2016-02-14:
Ok, this command works. I seem to be able to mount the share manually. Even after that the camera is still off. I understand it will be off until it can mount the share.

mount -t cifs -o username=security,password=pw //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security

mount -t cifs -o username=security,password=pw //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security

[root@meye-9ee35875 motioneye_10_10_0_149_securitycameras_security]# ls -la
total 84
drwxr-xr-x    2 root     root             0 Feb 14 13:11 ./
drwxrwxr-x    4 root     root          4096 Feb 14 13:21 ../
-rwxr-xr-x    1 root     root         73357 Oct 21 13:47 Credit Report.png*
[root@meye-9ee35875 motioneye_10_10_0_149_securitycameras_security]# 

but I think get this in the log

ERROR: failed to mount smb share "//10.10.0.149/SecurityCameras" at "/data/media/motioneye_10_10_0_149_securitycameras_security"

ccrisan, 2016-02-14:
Could you please enable debugging (from UI) and try again? The log might contain a few more pieces of information regarding the issue.

bazcurtis, 2016-02-14:
I assume this is the relevant part of the log?

2016-02-14 16:00:43: [motioneye]    DEBUG: updating SMB mounts
2016-02-14 16:00:43: [motioneye]    DEBUG: listing smb mounts...
2016-02-14 16:00:43: [motioneye]    DEBUG: making sure mount point "/data/media/motioneye_10_10_0_149_securitycameras_security" exists
2016-02-14 16:00:43: [motioneye]    DEBUG: mounting "//10.10.0.149/SecurityCameras" at "/data/media/motioneye_10_10_0_149_securitycameras_security" (sec=ntlm)
2016-02-14 16:00:43: [motioneye]    DEBUG: mounting "//10.10.0.149/SecurityCameras" at "/data/media/motioneye_10_10_0_149_securitycameras_security" (sec=ntlmv2)
2016-02-14 16:00:43: [motioneye]    DEBUG: mounting "//10.10.0.149/SecurityCameras" at "/data/media/motioneye_10_10_0_149_securitycameras_security" (sec=none)
2016-02-14 16:00:43: [motioneye]    ERROR: failed to mount smb share "//10.10.0.149/SecurityCameras" at "/data/media/motioneye_10_10_0_149_securitycameras_security"
2016-02-14 16:00:43: [motioneye]    DEBUG: 200 POST /config/0/set/?_=1455465640874&_username=admin&_signature=501cd4199083a4d2d92b485fc5ef9e9096cdb460 (10.10.0.149) 3477.16ms
2016-02-14 16:00:43: [motioneye]    DEBUG: 200 GET /static/img/motioneye-logo.svg (127.0.0.1) 5.14ms
2016-02-14 16:00:43: [motioneye]    DEBUG: recevied relayed event stop for thread id 1 (camera id 1)
2016-02-14 16:00:43: [motioneye]    DEBUG: clearing motion detected for camera with id 1
2016-02-14 16:00:43: [relayevent]    DEBUG: event successfully relayed
2016-02-14 16:00:43: [relayevent]    DEBUG: bye!

I've also attached the full log just in case
[me_log.txt](https://github.com/ccrisan/motioneyeos/files/129765/me_log.txt)

Thanks for looking.

ccrisan, 2016-02-14:
The sec option is probably the one that generates the problem. Could you please try passing the mount -t cifs command the option `-o sec=<mode>` where `<mode>` is one of:

 * `none`
 * `krb5`
 * `krb5i`
 * `ntlm`
 * `ntlmi`
 * `ntlmv2`
 * `ntlmv2i`
 * `ntlmssp`
 * `ntlmsspi`

See if any of these works.

bazcurtis, 2016-02-14:
Will do, but just to clarify it would look like this?

mount -t cifs -o username=security,password=pw,sec=xxx //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security

When I ran the command without the sec=, what would that have sent? I may not get a chance until next weekend now, but I will update asap.

Thanks once again.

bazcurtis, 2016-02-14:
I found this in the Mac system log if that helps.

Feb 14 16:05:16 Home-iMac-2 smbd[38233]: 10.10.0.153 SMB client not supported - Unicode, NT Errors, Long Names and Extended Security are required

ccrisan, 2016-02-14:
No idea what's used if you don't supply any sec.

bazcurtis, 2016-02-16:
I tried each one in turn and this was the output

[root@meye-9ee35875 media]# mount -t cifs -o username=security,password=pw,sec=none //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security
mount: mounting //10.10.0.149/SecurityCameras/Office on /data/media/motioneye_10_10_0_149_securitycameras_security failed: Permission denied
[root@meye-9ee35875 media]# mount -t cifs -o username=security,password=pw,sec=krb5 //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security
mount: mounting //10.10.0.149/SecurityCameras/Office on /data/media/motioneye_10_10_0_149_securitycameras_security failed: No such file or directory
[root@meye-9ee35875 media]# mount -t cifs -o username=security,password=pw,sec=krb5i //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security
mount: mounting //10.10.0.149/SecurityCameras/Office on /data/media/motioneye_10_10_0_149_securitycameras_security failed: No such file or directory
[root@meye-9ee35875 media]# mount -t cifs -o username=security,password=pw,sec=none //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security 
mount: mounting //10.10.0.149/SecurityCameras/Office on /data/media/motioneye_10_10_0_149_securitycameras_security failed: Permission denied
[root@meye-9ee35875 media]# mount -t cifs -o username=security,password=pw,sec=ntlm //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security
mount: mounting //10.10.0.149/SecurityCameras/Office on /data/media/motioneye_10_10_0_149_securitycameras_security failed: Invalid argument
[root@meye-9ee35875 media]# mount -t cifs -o username=security,password=pw,sec=ntlmi //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security
mount: mounting //10.10.0.149/SecurityCameras/Office on /data/media/motioneye_10_10_0_149_securitycameras_security failed: Invalid argument
[root@meye-9ee35875 media]# mount -t cifs -o username=security,password=pw,sec=ntlmv2 //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security
mount: mounting //10.10.0.149/SecurityCameras/Office on /data/media/motioneye_10_10_0_149_securitycameras_security failed: Invalid argument
[root@meye-9ee35875 media]# mount -t cifs -o username=security,password=pw,sec=ntlmv2i //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security
mount: mounting //10.10.0.149/SecurityCameras/Office on /data/media/motioneye_10_10_0_149_securitycameras_security failed: Invalid argument
[root@meye-9ee35875 media]# mount -t cifs -o username=security,password=pw,sec=ntlmssp //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security

[root@meye-9ee35875 motioneye_10_10_0_149_securitycameras_security]# mount -t cifs -o username=security,password=pw,sec=ntlmv2i //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security
mount: mounting //10.10.0.149/SecurityCameras/Office on /data/media/motioneye_10_10_0_149_securitycameras_security failed: Invalid argument
[root@meye-9ee35875 motioneye_10_10_0_149_securitycameras_security]# mount -t cifs -o username=security,password=pw,sec=ntlmsspi //10.10.0.149/SecurityCameras/Office /data/media/motioneye_10_10_0_149_securitycameras_security

Only these two seemed to work.

What does this mean?

On the Mac end a failed connection looks like this

SMB client not supported - Unicode, NT Errors, Long Names and Extended Security are required

a working one looks like this

SMB client not supported - Unicode strings are required

If I mount it manually and then set it up in the GUI the camera will go off. The motioneye log shows this.

2016-02-16 16:04:58: [motioneye]    ERROR: failed to mount smb share "//10.10.0.149/SecurityCameras" at "/data/media/motioneye_10_10_0_149_securitycameras_security"
2016-02-16 16:04:58: [motioneye]    ERROR: failed to unmount smb share "//10.10.0.149/SecurityCameras/Office" from "/data/media/motioneye_10_10_0_149_securitycameras_office_security"
2016-02-16 16:05:08: [motioneye]    ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_10_10_0_149_securitycameras_security/Office'
2016-02-16 16:05:08: [motioneye]    ERROR: failed to mount smb share "//10.10.0.149/SecurityCameras" at "/data/media/motioneye_10_10_0_149_securitycameras_security"
2016-02-16 16:05:09: [motioneye]    ERROR: failed to unmount smb share "//10.10.0.149/SecurityCameras/Office" from "/data/media/motioneye_10_10_0_149_securitycameras_office_security"
2016-02-16 16:05:09: [motioneye]    ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/media/motioneye_10_10_0_149_securitycameras_security/Office'

Thanks for looking into this.

ccrisan, 2016-02-17:
It looks like I need to add support for these security methods as well.

bazcurtis, 2016-02-17:
Is that hard? :)

ccrisan, 2016-02-18:
Nope, it's trivial.

Chiny91, 2016-02-18:
Ah good; could be useful to me also.   Thanks.

ccrisan, 2016-02-20:
Fixed upstream.

closed 2016-02-20 by ccrisan

bazcurtis, 2016-02-23:
Thanks for uploading the fix. When will it show up in the check for update option?

ccrisan, 2016-02-23:
You could either enable prereleases and it will show up right away, or you'll need to wait a couple of days before I can deem this version stable (or not).