motioneye-project/motioneyeos #162: Configure WIFI on Raspberry A+

Robin26, 2016-02-13:
Hello Mr Crisan,

My name is Robin, I'm french.

I recently bought a Raspberry Pi A+ with a 2A power supply, a camera module, and a WIFI dongle to build a WIFI surveillance camera.
I tried to install your OS motioneyeos but I have some issues.
First, I don't have any hub USB, thus I only can use one USB device.
I have written the .img on my micro SD card and powered ON my RPI A+, after plugging my keyboard device. I've got 
=> can't open etc/resolv.conf
Then, I tried to edit /etc/init.d/S35wifi in order to connect my WIFI dongle but I'm not familiar with "vi" (and my not adjusted french keyboard). Finally, I restarted my RPI A+ (unplugging the power supply) with my WIFI dongle without success.
I have some questions:
1. How shutdown motioneyeos (shutdown -h now doesn't work)
2. How configure my WIFI dongle (I saw your tweaks but I'm probably awkward)

I really thank you ! I need you.

Robin

ccrisan, 2016-02-14:
See this: https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Model-A.

In a nutshell: either use `writeimage.sh` with -n arguments, ran from a Linux machine, or prepare your OS on a Rasberry PI B1.

closed 2016-02-14 by ccrisan

Robin26, 2016-02-14:
Thank you for responding so quickly. I've got an other idea, can I use this an adaptator USB-RJ45 like this http://www.amazon.fr/Ugreen-Adaptateur-Ethernet-Support-Windows/dp/B00MYT481C/ref=cm_cr_pr_product_top?ie=UTF8 on my Raspberry PI A+ ?

ccrisan, 2016-02-14:
It probably will work. I couldn't tell for sure but normally that would be one way to proceed. Or you could simply buy a B+ :)

Robin26, 2016-02-14:
Yes but as I'm student, my wallet for this project is limited. I've got my RP A+ for 30 euros with shipping, on kubii.fr, I should have chose the model 2 for 40 euros...
Please could you help my to make the best choice ?
1. Return the model A+ and buy a model A+ (=> lose 5 euros shipping + ~5 euros returning).
2. Keep my model A+ and buy an ethernet adaptator for 10 euros.

It's only to run your OS motioneyeos with a module NOIR camera and send 2 pictures/second on a server ftp, so do you think the 1. worth it (with 256Mo RAM) ? Or I should move on the model 2 in choosing the option 2. ?

Thank you a lot !