motioneye-project/motioneyeos #163: Gogle Drive create/use folders

AdamMalkin, 2016-02-13:
Hi ccrisan,
Love your os, just setup my first camera and plan to setup 4 more over the next few weeks.
I have google drive setup and working to a folder named /CCTV/ but wanted to use date folders like local storage e.g. /CCTV/%Y-%m-%d/
If i set the movie name as %Y-%m-%d/%q_%H-%M-%S then only the %q_%H-%M-%S is used in drive.
If i set the folder in the google drive setup as /CCTV/%Y-%m-%d/ i get an error in the log saying folder /%Y-%m-%d/ does not exist
I tried creating the %Y-%m-%d folder manually, but motioneye still puts the files in the root folder

is there a way for me to get this working?

ccrisan, 2016-02-14:
Not really. I know things with Google Drive and Dropbox are not currently ideal, but I didn't have the time to take care of them properly. I'll see what I can do about your issue.

AdamMalkin, 2016-02-14:
That would be amazing, thanks
Its ok with one camera, but with a few the files would take a lot of maintenance.
In the meantime i can setup some scripts on another PC to tidy files up so no rush :)

bortek, 2017-03-01:
Please check the latest version as this might have been implemented. Otherwise post your question on the project forum and discuss it there. https://groups.google.com/forum/#!forum/motioneye

closed 2017-03-01 by bortek