motioneye-project/motioneyeos #164: writeimage.sh makes the system hang

izmailoff, 2016-02-14:
If I'm not logged-in as `root` and run `writeimage.sh` this line

`test "root" != "$USER" && exec sudo $0 "$@"`

will case a fork bomb and will fairly soon make the OS hang completely. I'll submit a pull request to address that.

Maybe it's also not a good idea to run it as `sudo` without asking the user. A more common approach is just to say: "please run as root", so a user would know the implications.

Thanks

ccrisan, 2016-02-14:
Yeah, I'm not really sure why that line was there. I admit the best thing to do is let the user explicitly run it as root. However I fail to see how this could turn into a fork bomb, unless your sudo command doesn't do what it's supposed to do.

closed 2016-02-14 by ccrisan

izmailoff, 2016-02-15:
Indeed, what's in `$USER` depends on the system. I'm using latest Fedora (23) and get the same value in `$USER` whether I'm a root or not, which causes infinite loop. I haven't had a chance to check on other Linuxes and alike.