motioneye-project/motioneyeos #167: Broken UI on Chrome

hamid-elaosta, 2016-02-15:
Chrome: Version 45.0.2454.99 (64-bit)
OS: Linux Mint 17.3 "Rosa"

All drop-down boxes show an empty white box. They do work on Firefox.

In chrome, I can use up-down keys to switch the value in the drop-down, but if I open the drop-down it is always just a white box.

Gordon-1, 2016-02-24:
Chrome (version 48.0.2564.116) works fine with MotionEyeOS on my Win 10 box.. Are you sure you have the latest version of Chrome?  Google tells me this (emphasis mine):
_The Stable channel has been updated to 48.0.2564.116 (Platform version: 7647.84.0) **for all Chrome OS devices**._

hamid-elaosta, 2016-02-24:
I'm sure I don't have the latest version. I have 45. Chrome notified me today (funny coincidence) that 48 is available.

I'll test tomorrow when I return to the office using 48 and let you know.

ccrisan, 2016-02-24:
I'm experiencing the same issue with many other websites/webapps that customize comboboxes appearance in any way. I'm not sure how to work around this.

donjakobo, 2016-03-03:
I'm also not seeing the SVG icons come up over the camera (when you click on a camera). Using latest updated Google Chrome on Win7.  RaspberryPi 1 B model.  Camera works but no web overlay buttons on the camera

[image]

ccrisan, 2016-03-03:
Any 404s in the dev tools console?

donjakobo, 2016-03-03:
> Any 404s in the dev tools console?

I found something interesting, when I call the motionEye RPI box on the local private IP 10.0.0.X, it doesn't show the icons, but using an external domain pointing to my home network and using a port forward, I see the icons fine:
[image]

IP accessed, image doesn't even show up/load? SVG over local ip issue?
[image]

There are no 404's being thrown. The only difference I see is the port forward of an external domain to the motionEyeOs box.

ccrisan, 2016-03-03:
That's terribly strange. Could you please try clearing your cache?

hamid-elaosta, 2016-03-03:
I tested with latest chrome and the drop down boxes are working fine.

I don't recall whether the overlay icons were present or not.

donjakobo, 2016-03-03:
I just cleared cache AND tried in incognito mode, here is the local ip view, no 404's again:
[image]

ccrisan, 2016-03-03:
I assume it doesn't happen in Firefox or IE, does it?

donjakobo, 2016-03-03:
I tested and icons work in IE, I don't run FF so can't try, but this seems isolated to Chrome on a local IP.  I'm scratching my head on this one, must be some chrome defect svg on private ips? Maybe someone else can replicate.

bortek, 2017-03-01:
I dont see this problem using latest Chrome and latest MotioneyeOS. I will close this one for now, reopen if you are still seeing it.

closed 2017-03-01 by bortek