motioneye-project/motioneyeos #168: RPI A+ surveillance system

Robin26, 2016-02-15:
Hi Mr Crisan.

I would like to give you a feedback on my first issue https://github.com/ccrisan/motioneyeos/issues/162
Finally, I used your bash script on my linux laptop (dual boot W10 + eOS) and with -n arguments => which allowed me to rewrite correctly my SD card. It was a success ! ;)
Now I'm able to see my camera in my browser, and I'm actually impressed by your UI motioneye (I'm web developper during my free time), responsive, clear, and beautiful !

Just a last question before it would be perfect : I use the Sainsmart camera noir http://ecx.images-amazon.com/images/I/41Hzy8lPr1L.jpg but the 2 included IR LEDs fixed on this camera are always ON (red color). Could I disable them the day with your OS ?

Thank you a lot Mr Crisan !

I'm looking forward to hearing you.
Robin

ccrisan, 2016-02-15:
I don't think those leds could be disabled programmatically. They are automatically controlled by the light sensors. Even if they were controllable programmatically, there's no support for such functionality in motionEye(OS).

closed 2016-02-15 by ccrisan

mano1979, 2016-12-09:
I use the same IR spots an mine are blinking on and off all the time. Even when i capture video.
It didn't used to do that before. Maybe i installed a wrong driver? 
I installed the default driver in raspi-config.