motioneye-project/motioneyeos #169: Addind Cameras Manually

n3rxstech, 2016-02-18:
How do I manually add cameras to MotionEyeOS? When it was motionpie there was an option drop down at the top of the page but in MotionEyeOS I see no such option anywhere when logged in as Admin.

ccrisan, 2016-02-18:
Make sure you are indeed logged in as admin. You'll then see the menu icon on the upper-left corner. Open it and click on the cameras combo box. You should then see the add camera option.

n3rxstech, 2016-02-18:
There is no cameras Combo box sir. I am including a screen shot.
[https://www.dropbox.com/s/rtt201759k2rseg/Screenshot%202016-02-18%2007.35.50.png?dl=0](url)

ccrisan, 2016-02-18:
You have enabled [Fast Network Camera](https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera) which doesn't allow adding more than one camera.

closed 2016-02-18 by ccrisan