motioneye-project/motioneyeos #170: system always stop after a few days

alainthebest, 2016-02-18:
I'm working with a Pi2, three axis cameras and the motioneyeos image. I record on the SD card (32GB) 
because recording on usb give problems. All works maybe for days with two cameras and just one day with three cameras. All just stop at a time. When I reset all, that work again for a few time. I join my logs recorded just after the last reset..
Can someone see something wrong ?

thank's

[log02.zip](https://github.com/ccrisan/motioneyeos/files/135840/log02.zip)

Chiny91, 2016-02-18:
I have exactly the same issue with my Pi2 and 3 IP cameras and have done since the Dec 15 (I think) update.   It looks like a motion-mrdave wobble to me, in my logs, at the moment.   Certainly I had no problem at all for 2 months last Sept/Oct whilst I was away with no access to the cams.   Occasionally motioneye wobbles but that appears to reboot itself, presumably under the watchdog, although it did fail once.

I'm quite tempted to extract the earlier motion daemon and try that in the current system.

But, comments by others would be very interesting before I break my setup.

ccrisan, 2016-02-18:
Things are a bit unstable with netcams, as far as I can tell. Unfortunately that's a motion issue and I  won't touch its source code in hope of fixing problems. The problem seems to manifest especially when dealing with RTSP cameras but users have sent logs with non-RTSP IP cameras where threads would hang and thousands of lines of netcam-related errors were visible.

I can say that using motion-mrdave with 7 motionEyeOS-based netcams has been working perfectly for me so far.

reddai75, 2016-02-20:
@ccrisan what hardware are you using ? I mean what is your configuration ?

ccrisan, 2016-02-20:
I have 7 (and counting) Raspberry PI B+ with official CSI cameras running motionEyeOS in FNA mode. An Odroid XU4 has all of them added as network cameras and does the motion detection job using MrDave's motion version.

mikecarr, 2016-02-20:
I have also found clearing images and videos out fixes it as well.

alainthebest, 2016-02-21:
I use 3 Axis 221 cameras with the last firmware. So I've moved the power supply in an other plug to have more space for heat dissipation and the PI2 is also in an "open box" with heatsink on the two chips. Quality of video and pictures are 100% for 2 frames/sec. It's now 3 days without problems.
Dear Ccrisan, it's a great job, thank's !

closed 2016-02-21 by alainthebest

Chiny91, 2016-02-21:
Poking about, I discover that the "old" motion daemon remains as motion-svn, together with an interesting bash script, /usr/bin/motion.   So, I have hacked this script such that starting line 28, I now have:

else # two or more cameras
  # Hack this to force the old motion daemon
  # motion=$MOTION_MRDAVE
motion=$MOTION_SVN

This is obviously crude but appears to work although I may have missed something.   The point is to check whether motion-mrdave is the cause of my problems by being less stable than motion-svn.   motion-mrdave rarely lasts more than 1 days *for me*.

Don't try this unless you want to break things.

ccrisan, 2016-02-21:
Indeed, motion-svn is still available but not used by default. It's not really the "old" stable version, but some intermediary version taken from their official subversion.

Chiny91, 2016-02-21:
Poking about, I discover that the "old" motion daemon remains as motion-svn, together with an interesting bash script, /usr/bin/motion.   So, I have hacked this script such that starting line 28, I now have:

else # two or more cameras
  # Hack this to force the old motion daemon
  # motion=$MOTION_MRDAVE
  motion=$MOTION_SVN

This is obviously crude but appears to work although I may have missed something.   The point is to check whether motion-mrdave is the cause of my problems by being less stable than motion-svn.   motion-mrdave rarely lasts more than 1 or 2 days *for me*.

Chiny91, 2016-02-24:
Update after 3 days:
Using _motion-svn_ rather than _motion-mrdave_ has partly resolved the issue.   I have had no locking-up ("crashing") of motionEyeOS.   _motion-svn_ appears to generate just the same amount of log file, in terms of _info_ and _warning_ but no error so serious, that motionEyeOS cannot recover (reboot).   motioneye reboots fairly often, typically approximately hourly but with no discernible pattern.  I can see the occasional loss of network (can't connect to google.com) as well.

I suspect assorted causes but tbh, so long as motionEyeOS can recover from being locked-up (as it has so far), I'm not too concerned.   My R Pi 2 takes 42 seconds to reboot, so the loss of  security in that time is limited.

For those that want to try _motion-svn_, there is a suitable place in the GUI.   Just go to _Expert Settings_, _Motion Binary_ and change _/usr/bin/motion_ to _/usr/bin/motion-svn_.