motioneye-project/motioneyeos #171: RPI 1 B reboot issue with increasing Frame Rate and Resolution

clawwd, 2016-02-19:
I am having issues with my PI rebooting once I starting bumping up the resolution, frame rate, etc beyond 
the lowest settings and I am wondering if its the PI or the software. I am actually leaning toward thinking its the 20' USB cable I am using and a power issue at that length or something to do with the CPU overloading. It just reboots over and over. I am using the default settings other than turning the resolution down to the lowest setting and frame rate to 2.

My setup
RPI 1 model B
Microsoft Lifecam 3000
20' USB power cable
USB WiFi dongle

ccrisan, 2016-02-19:
You have already done part of the work yourself by identifying possible causes. Here are a few ideas:

 * make sure you have not overclocked your PI
 * if you're indeed using a B and not a B+, try bypassing the polyfuse which seriously limits the max USB drawn current
 * try using an ethernet connection instead of the wifi one
 * try using a powered USB hub

clawwd, 2016-02-19:
I might try replacing the power supply with a larger one. Any suggestions?

clawwd, 2016-02-22:
Shortened the cable to 10' which fixed the reboot.

closed 2016-02-22 by clawwd