motioneye-project/motioneyeos #172: Web interface unaccesible after changing port

jimmyeao, 2016-02-20:
Changed the port to 83, now it doesnt repsond on that port, ssh in and do a netstat and it shows time wait. Scan with fing and port 83 is not showing. Are there iptables in use?

ccrisan, 2016-02-21:
Nope. iptables is not used by default. Were you trying to access the system directly from your local network? Would you be able to recover some log and config files from your SD card to see what happened?

jimmyeao, 2016-02-21:
Sorted it now, had it set to medium over lock (pi b, 512mb) and what was happening was a constant reboot- reflashed it and set the over lock to low and it all seems stable now..

Thanks!

Jimmy White

closed 2016-02-21 by ccrisan

emosbek, 2017-02-09:
Hello,

Same happen to me just now. connected the r.pi to a monitor and are up and running via terminal. Any chance for me to revert port change?

Regards