motioneye-project/motioneyeos #174: Save Fast Network Camera Streams?

tsrgit, 2016-02-21:
I am trying to set up the "Multiple devices with a central server" scenario described [here](https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios).  I am running motioneyeos 20160109 on a RPi B (Pi NoIR camera module) that is configured as a Fast Network Camera, and I have motioneye 0.30 running on an Ubuntu server.  I can use my web browser to connect to motioneye on the server and view the MJPEG video stream from the RPi just fine.  But I can't figure out how to get motioneye on the server to store the video streams for later viewing.  The "Scenarios" page says that "the media files can be saved on a hard drive directly attached to the server."  But the "File Storage" tab in configuration page of motioneye has no options for where and how to save media files.  It only presents a single option to "Run A Command" after a media file is created.  I checked motioneye.conf on my server, and the media_path is set to the default location of /var/lib/motioneye.  My understanding is that by default motioneye runs as root.  But just in case I went ahead and set the permissions on /var/lib/motioneye to rwx for all.  So far, no files have been created in this directory.  What do I need to do to get motioneye to save MJPEG streams from network cameras?

ccrisan, 2016-02-22:
Are you positively sure that the "File Storage" section has a "Run A Command" option? May I see a screenshot?

tsrgit, 2016-02-24:
I think my problem was that when the remote RPi camera was originally added to the server instance of motioneye, it was configured on the server as a "Simple MJPEG Camera".  The File Storage tab looked like this:

[image]

I deleted the camera from the server and re-added it as a "Network Camera" streaming on port 8081.  Now the configuration panel on the server shows the entire set of motion options, including these file storage options:

[image]

I can now save the video stream to the server!

ccrisan, 2016-02-24:
The bug report might however be valid, as the Simple Network Camera mode should never display a "File Storage" section at all.

closed 2016-03-03 by ccrisan