motioneye-project/motioneyeos #175: Streaming image frame rate drops/dies when movement is on the camera

clawwd, 2016-02-22:
I am noticing somthing happening on both of my cameras that wasnt happening with motioneye. The streaming image frame rate drops/dies when movement is on the camera. You get the immediate portion when something shows up on the camera followed by a lag then the image stalls altogether for a couple of minutes. I have turned down frame rates, image qualities, email notifications off, etc. 

The camera is not rebooting or going offline because I have a constant ping going which never drops as you would see when a camera is rebooting.

Before I switch back to Motioneye, I wanted to see if you all had any thoughts on what I may be doing wrong.

clawwd, 2016-02-22:
Setup:
Raspberry Pi Model B REV 2
Microsoft LifeCam HD-3000
Ralink RT5370

ccrisan, 2016-02-22:
Please attach your log files.

clawwd, 2016-02-22:
Here are the log files
[Archive.zip](https://github.com/ccrisan/motioneyeos/files/140724/Archive.zip)

clawwd, 2016-02-22:
Here are the logs
[Logs.zip](https://github.com/ccrisan/motioneyeos/files/140733/Logs.zip)

ccrisan, 2016-02-22:
It looks like the motion daemon hangs sometimes and needs to be restarted. It may be the case of a buggy camera driver. Please try using a different camera and see if the problem persists.

One more thing: older RPi models have those horrid polyfuses limiting the USB current. If that is your case, you might want to upgrade your PI board. You may also try a powered USB hub.

clawwd, 2016-02-22:
Thanks for the quick reply.

Is there a way to update the driver? I have not seen this on motionpie.

ccrisan, 2016-02-22:
No idea. I would start by ruling out power-related issues. motionEyeOS might be just a bit more power hungry.

clawwd, 2016-02-24:
Closing this. Moved back to motionpie

closed 2016-02-24 by clawwd

ccrisan, 2016-02-24:
@clawwd it may be a problem with MrDave's motion. Could you please try setting Motion Binary to "/usr/bin/motion-svn" in Expert Settings and see if that fixes your problem?

clawwd, 2016-02-24:
It helped a little bit.I may have captured a few more frames but in the end it stalled again.

Appreciate your help