motioneye-project/motioneyeos #176: logrotate

Chiny91, 2016-02-22:
I've noticed that the various logs in _/data/log/_ are getting very large, especially _messages_ (58 MB) and _motioneye.log_ (23 MB).   Both of these logs go back to 26/12/15, probably when I installed motionEyeOS - in other words, no logrotate.

Have I broken something ?

ccrisan, 2016-02-22:
There is no logrotate indeed. I'd probably need to think of a way to trim those logs.

Chiny91, 2016-02-22:
Is there a reason not to use _logrotate_ ?   It is long-established so stable, although it does come with a lot of features (example, compression) which will increase the overall size of motionEyeOS which may be undesirable.

Personally, I'd be happy with renaming a week of logs and deleting the following week but it is 100% certain that someone will want more features :astonished:

ccrisan, 2016-02-23:
@Chiny91 although I use logrotate on all my production servers (not related to motionEye), I don't think it would be of any use on a camera device. If you think of motionEyeOS as a camera device firmware, there's definitely no reason why system logs should be kept for more than let's say a month. They are purely debugging logs.

As a consequence, I believe trimming them on a regular basis would be more than enough.

Chiny91, 2016-02-23:
_no reason why system logs should be kept for more than let's say a month_ - now there is a challenge  :pensive:   Hard-to-access systems could find it useful, rarely checked systems etc.   Incidentally, all my camera devices can keep logs which I ignore (usually) just like my ntp server logs, DNS server logs unless there is a problem.   Of course, I would not have been checking motionEyeOS logs if I had not been suffering too many occasions of the system locking.   I guess that is the reason for _logrotate_, occasional investigation of problems which may stretch back over time.   Some users may want the flexibility.

For yourself as an active developer, the issue is a little different.   Firstly it is unlikely your system will be left on its own for long and secondly, users reporting problems to you are likely to be near their systems at the time so you will get recent logs.

As an aside, that second item has probably led to the default log level for motioneye being _info_.   I have found that a little too verbose for myself, especially bearing in mind I wore out an SDcard last year so have cut the level to _error_.    I guess this just shows, in general, a need for flexibility.

Just my 2p worth.   I'm sure others will have their own views.

ccrisan, 2016-02-24:
I get your point and you're right. There are some cases where extended log information is required. However I am doing my best to keep the OS as light as possible. Therefore I guess trimming files based on a configurable duration would be the best option. I'll try to see how logrotate would affect the OS footprint and everything but I believe that users like you could simply tweak the duration or totally disable the trimming, provided there will be such an UI option.

ccrisan, 2016-06-12:
c4165ac02ef91d7f83acf27cb89ecb34f08ed4a7 adds logrotate support.

closed 2016-06-12 by ccrisan

Chiny91, 2016-06-17:
Ah... just back from 2 months away and noted the addition of _logrotate_ whilst perusing the repository.   _dmesg_ had reached 185 MB whilst away :anguished:    Good addition.