motioneye-project/motioneyeos #178: Exposure setting for the Raspberry Pi 2 camera board

Priso19, 2016-02-26:
Hi,

First, thank you for your great development, amazing job, really.

I looking for the exposure settings of my Raspberry Pi 2 camera board and I can't find them in the left admin menu. So I make a SSH connection without any results, where are they? Is there a config file somewhere?

My camera is indoor and I need the "night mode" setting all the time.  

Thanks for your help.

ccrisan, 2016-02-27:
Extra RPI camera-specific settings are available only in Fast Network Camera.

closed 2016-02-27 by ccrisan

cooleyandy, 2017-01-11:
Hi Calin,

I understand the ISO settings are only available in the Fast Network Camera setting.
Just curious why that is from a technical perspective. Is it an api/library limitation, or any inherent reason it shouldn't be done?

I've tested using the Fast Network Camera at ISO 800, and the resulting photos are so much brighter than the regular camera settings.

ccrisan, 2017-01-11:
motionEye normally uses motion as a backend and v4l2-ctl for controlling things such as brightness and contrast. Now, ISO and other parameters might be available via v4l2-ctl when a RPi cam is installed, but they aren't available in general and therefore motionEye doesn't generally implement them.

With FNC, things are different, as motionEye uses streamEye + raspimjpeg.py as backend and things are a bit different. Everything with FNC has been developed specifically for the RPi cam and thus I could implement every single available setting, no matter how particular it was.

cooleyandy, 2017-01-11:
Oh I see. So motionEye was developed with compatibility with other USB cameras in mind, so the available settings would make sense for all cameras, and not specifically RPi cams.
Thanks for this tidbit of insight.