motioneye-project/motioneyeos #179: trouble using write.sh

cwwells, 2016-02-27:
since I am a novice, I reviewed all the posted issues and googled lots of linux stuff, yet no success..Questions are:
- I partitioned sd card to sdb1 and formatted to fat 32  --is this the right starting point ?
- re the script to run write.sh, ./writeimage.sh -d /dev/mmcblk0 -i "/path/to/motioneyeos.img" -n "yournet:yourkey"  I replaced mmcbllk0 with sdb1 (right ?)  changed to my path for "/.../motioneyeos.img"  --keeping the parentheses (right ?)   --I also kept the parentheses on  the login --"myssid":"mykey"  (ok ?)
Symptoms are can't connect and can't mount the sd card in the laptop file manager

Thanks, Curt

/

ccrisan, 2016-02-27:
Don't bother formatting anything. writeimage.sh does everything for you. Use /dev/sdb instead of /dev/sdb1.

cwwells, 2016-02-27:
Wonderful !! it works !!  Thanks so much Calin !!

closed 2016-02-27 by ccrisan