motioneye-project/motioneyeos #180: Day & Night Changes

antares2001, 2016-02-27:
Hello,

first of all thanks for your amazing work. This is a fantastic tool.

Just a minor question: could you guys add a feature that would - based on a given time by the user (or based on sunrise / sunset from a geo-coordinate-based service like here http://www.date-and-time.net/) - change the settings of the camera, so that, e.g., the exposure mode is set to night at sunset and iso to 800 and to a lower exposure and iso 200 at sunrise.

Thanks again! Great job, guys.

5tormChild, 2016-02-28:
Yeah, that would be nice!
You don't need to use an external webpage though, here is an example of daytime calculation:
https://www.reddit.com/r/raspberry_pi/comments/29w58y/sharing_my_timelapse_script_with_sunrise_and_dawn/
"Note: the script uses a python module named "Astral" https://pypi.python.org/pypi/astral/" - here is a possible and easy solution; just set the location (lat/long), and you can calculate the exact time of dawn and dusk, and you would need to implement only one checkbox (plus the location settings) to avoid pitch black night images.

antares2001, 2016-02-28:
I'm also looking a bit self into the OS to understand how it works. I found that in raspimjpeg.conf on /data/etc the settings of the camera are put. Yet, I'm not sure what I need to do once I change settings in that file. Is it a simple service reload / restart or something different?

ccrisan, 2016-02-29:
The `raspimjpeg.conf` file is used only when Fast Network Camera mode is enabled. In that case, `streameye.sh restart` should be enough when changing those parameters.

antares2001, 2016-02-29:
Hi @ccrisan and thanks for the reply. I'm trying that out.

Yet, I found that there's some difference to the behavior I had on previous installations where I was using raspivid with ffmpeg or u4vl when making the settings. I was able to get much better (i.e. brighter) video during night when setting exposure to night or verylong and left ISO set to 0 (it seems to me that there is an auto mechanism that then sets ISO even higher, because the picture was brighter than with ISO 800), but I don't know what exactly happens.

On MotionEyeOS I notice, that it makes de-facto no difference on the RPi 2 with the Pi-NoIR cam whether put exposure to auto or night. It's always the same. Do you have an idea why this happens?

Also - when putting DCR to high, the camera stops working, while low and medium it works (and seems to make the scene somewhat brighter).

Do you have any idea, how I could achieve better (longer) exposure and brighter images with Fast Network Camera at night? Clearly, the FPS would drop then, but I'm happy with 7-10 FPS at night, while during day I get 25 FPS, which is fantastic (Overclock Raspi 2 mode and 512 MB GPU RAM).

I also read on other threads about the MPEG-license that one can acquire for the Pi. Will that make any difference? How and where do I enter the license key on MotionEyeOS? Will that make it possible to directly stream video (h264/mpeg) instead of MJPEG?

Thanks again for your amazing work.

ccrisan, 2016-02-29:
Regarding the RPI camera parameters, everything that's exposed via [picamera](https://picamera.readthedocs.org/en/release-1.10/) is available in the UI. I have no idea how they affect anything behind. I myself use the original CSI camera module without any tweaks and haven't had any serious ISO/brightness related problems with the default settings. The GPU acceleration is not used by motion, so don't bother buying a mpeg license or anything like that.

fastjames, 2016-07-04:
Related, this poster created a script to do what's under discussion here. Unfortunately the script link on his post is broken: https://blog.ligos.net/2016-04-18/Day-Night-Cycle-For-MotionEye.html

closed 2017-03-01 by bortek