motioneye-project/motioneyeos #181: Unable to SSH to MotionEyeOS

rakeshprabhuvj, 2016-02-28:
Hi,

I'm not able to SSH in to MotionEyeOS.

I tired using admin as well as root as user name and the password I had set in Web GUI, but always it says Access denied.

Please let me know how I could SSH in?

ccrisan, 2016-02-28:
Please attach your boot.log file (downloadable from the UI).

boomstick8x, 2016-02-28:
I have similar problem, but I cant even login into UI...
Tried to remove X from passwd file and remove shadow and shadow- files but no luck. Found in the boot.log line
'''
 * Setting root password: done
'''
But how to totally reset password?

rakeshprabhuvj, 2016-02-29:
Please find the boot.log attached.

[boot.txt](https://github.com/ccrisan/motioneyeos/files/150270/boot.txt)

ccrisan, 2016-02-29:
The log appears to be fine. Do you have any special characters in your admin password? Please try changing it to a simple one and see if that works. Then, please report back the special characters so that I can escape them.

rakeshprabhuvj, 2016-02-29:
I updated my password to numeric as I had read on one of the forums this
could be the reason.

But this did not help me as I`m unable to login via Putty

Thanks,

Rakesh

Mob: 9846202053

ccrisan, 2016-02-29:
This is really the first problem of this sort. Are you 100% sure you're actually connecting to the correct IP/machine? Is there any text that shows up when you're prompted for password? Is it of the form meye-xxxxxx?

rakeshprabhuvj, 2016-03-01:
Yes. it says meye-ef9f6439
[image]

ccrisan, 2016-03-01:
I'm sorry but I'm out of ideas. Could you please try writing the OS image from scratch and repeat the procedure?

stavropn, 2016-04-25:
I have the same problem. What is the password at this point?

al090187, 2016-04-25:
Might be wrong, but if the password is numeric and you are using the numeric keypad, check your Num lock status. I have tried entering a PIN many a time and ended up moving the page around instead. And, of course, with the password not being visible you wouldn't know.

Ethocreeper, 2016-04-27:
for me its working fine.
for username i use the admin and my password

bsilva-bentlabs, 2016-09-12:
I had this issue, but it went away when I freed up some disk space.

PubuduAravinda, 2016-10-14:
Hi,
first of all thank you for grate work. 
I need to configure DDNS server for MotionEye. but it is unable to give "SUDO" access to my Motion OS.
I checked the all of file, there is no way to change the settings as 'Root' login.
Please help me.
Thank you again.

ggramlich, 2016-10-16:
I am not sure, if this is the best place
@ccrisan  I could open another issue, if you want me to.
I had the problem, that I could not ssh into the Raspberry PI 3 with the internal wifi adapter, but it worked via (cable) lan. I used a fresh install of MotionEyeOs with the 20160919 pi3 image.
The ssh login hung after entering a correct password (denying access for a wrong password immediately).

I found this thread
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=138631&start=100
And followed the suggestion that came up at the end of page 4 of the thread and was confirmed multiple times on page 5.

I added the line

`IPQoS cs0 cs0`

to /etc/sshd_config

Now I can also login via wifi.

ccrisan, 2016-10-16:
Please open a new issue so we can keep track of it. I'll have to investigate this - I must admit I have never seen this option.

ccrisan, 2016-10-26:
(Hopefully) fixed with d2dfeb9481fc625d25bfc4b97954a5ae4aae61b9.

closed 2016-10-26 by ccrisan