motioneye-project/motioneyeos #182: Raspberry PI Zero

rockman55, 2016-02-29:
Having an issued trying to write initial image to Raspberry Pi Zero. I have used both versions of the software   (20151103 and 20160109 motion eye). All I get is the rainbow screen at the beginning and nothing else happens. I have been using this software for some time on my Raspberry Pi 2 Model B with no problem. I would like to be able to use this Pi Zero with a USB webcam. Have seen some other request and posts about the Pi Zero but not sure if anyone has got it to work. Thanks

ccrisan, 2016-02-29:
I haven't gotten my hands on a RPI0 yet. Sorry.

hxpaul, 2016-05-19:
I have just installed this on a raspberry pi zero, install went smoothly using motioneyeos-raspberrypi-20160410.img and motioneye is running, but it does not recognise the camera. This is all new hardware today, there might be some other issue, I'll do some experimenting.

hxpaul, 2016-05-19:
Pi Zero needed a firmware upgrade so I guess that needs a new build of motioneyeos

ctrunk514, 2016-05-26:
Just an FYI in case anyone finds this post and has a similar issue.  I had Motioneye working perfectly on a PI3 with wifi and it was awesome. I took the micosd card out and threw it in a RPI0 and motioneye could not detect the camera.  Unable to open video device was all I could get.  I ran sudo apt-get dist-upgrade and bingo, the camera was detected and motion eye started working.  Hope this helps someone.

kalgraff, 2016-05-27:
I try to run sudo apt-get dist-upgrade on MotionEyeOS, but it dos not work. Are you using MotionEyeOS or MotionEye?

rockman55, 2016-05-27:
You cannot use apt-get dist-upgrade with MotionEyeOS. The only way to update is through the user interface.

Danny

doughadfield, 2016-05-29:
I've just received my pi zero with camera cable, and I can confirm that motioneyeos doesn't work with the (original) pi camera on the zero board. Using a Raspbian lite image and mjpg-streamer (after updating firmware with rpi-update) the camera works fine, so it's definately the motioneyeos image that's the issue (using the latest). I would be happy to help with any beta testing.
Thanks again for this superb software!
Doug

doughadfield, 2016-05-29:
btw, I tried copying the firmware file from the /boot filesystem of the raspbian image to the motioneyeos image, but that doesn't work (or I'm doing it wrong).

ccrisan, 2016-05-29:
I'm working on it.

raspberrycoulis, 2016-06-01:
How does installation work on a PiZero if it needs a network connection, but not LAN port? Can you use the wpa_supplicant trick (explained here: https://www.raspberrypi.org/blog/another-update-raspbian/) to preconfigure your WiFi network prior to first boot? Or is there another way to configure the WiFi without a physical LAN connection?

ccrisan, 2016-06-01:
I have just updated the [troubleshooting page for RPI model A family](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-(Plus)-And-Zero) to include instructions on how to preconfigure wifi with /boot/wpa_supplicant.conf.

reggiekush, 2016-06-03:
@ctrunk514 Do you have a problem with it crashing on you? I've got my camera running at the lowest resolution and motioneye will crash as soon as I pull it up on a browser.

ccrisan, 2016-06-05:
A new OS image with Zero and camera v2.1 support will be available later today.

closed 2016-06-05 by ccrisan

RudyRipDraw, 2016-08-11:
Hi,
For the RaspberryPI zero, I downloaded the image onto and SD and booted.
My PiZero has HDMI display, so I can watch what is going on, usb hub to keyboard/mouse and wifi dongle.  I know the wifi needs to be configured after booting, but according to the HDMI display, the PI zero is looking for a "network wired" interface, the reboots and retries.

Any help on what I'm doing wrong would be greatly appreciated.  BTW I got motioneye to work on the PI both on wifi and ethernet.

thanks

ccrisan, 2016-08-11:
Can you please attach your log files here? Normally the OS requires a network connection, whether it's wired or wireless.

fersd, 2016-08-11:
I'm going to try it on a pi zero. The highlighted link is for a pi zero correct?
[image]

ccrisan, 2016-08-11:
Yes, that's the one.

RudyRipDraw, 2016-08-11:
Let me start by just thanking you for all the work you have done on this and also your efforts to support this for people like me.

Just so that there is no confusion on the thread.

[cid:image004.png@01D1F3A9.A5BD1530]

This is photo of the HDMI boot screen, after the last line, it retries.  I hope this was the info you were asking for.  If not, let me know.

[cid:image002.png@01D1F3AA.098424A0]

RudyRipDraw, 2016-08-11:
[image]

RudyRipDraw, 2016-08-11:
Sorry, new to communicating vi Git,    

After not finding the non-existent wired network, it reboots. 

I do have a powered USB-HUB--> wifi doggle, keyboard/mouse

thanks 

[image]

[image]

ccrisan, 2016-08-11:
@RudyRipDraw have you configured the wpa_supplicant settings as indicated [here](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-%28Plus%29-And-Zero)?

RudyRipDraw, 2016-08-11:
Calin,

Thank you for the response.  I've moved the SD over to the Pi B and have connectivity to the Ethernet.

Sorry, but I just a little more help on the config of the wifi.

Specifically where should I put the wpa_supplicant.conf file with my ssid and password?

[image]

[image]

RudyRipDraw, 2016-08-11:
just need to config wifi... :-) 
everything else is working on the B

[image]

RudyRipDraw, 2016-08-11:
or I don't know how to correctly invoke the write.sh script

[image]

ccrisan, 2016-08-11:
You need to choose *one* of those methods, not all of them :)

If you have managed to boot the image on a RPi model B, just enable Advanced Settings and, under the Network settings section, enable and configure your WiFi.

If you wanted to go by providing your own wpa_supplicant.conf, you should have placed it on the boot partition using your *Windows* machine (since you probably wouldn't have had access to a Linux machine).

RudyRipDraw, 2016-08-11:
Calin,

Thank you for your help.  It is all working.  I made some rookie moves, so let me summary my steps so that perhaps the next person can leverage this thread instead of your time.

1.  The Pi Zero runs the same image as the Pi Model B, so if you have a Pi Model, start with that unit because it has Ethernet

2. Install the wifi dongle and wired Ethernet

3. Boot the Pi Model B and find out the ip address from the router

4.  Point browser to ip address of Pi Model B from another machine on the same lan

5.  Login as admin, which defaults to no password
[image]

6. Change Advance Setting to ON otherwise you will not see the Network options
[image]

7. Under Network, enter the Wireless Network Name and Wireless Network Key,  then click the Apply
[image]

8. System will reboot.  But power off the Pi Model B and move the SD and wifi dongle to the Pi Zero 

This works for me!

thanks

ccrisan, 2016-08-12:
Perfect! Thanks for sharing this.

pjockey, 2016-08-20:
I just downloaded the 20160801 image and all I get is the square rainbow screen and nothing else. I've given it 5 minutes, but nothing. I'm running a RP 3 with a SainSmart Infrared Night Vision Surveillance Camera. I've played with some Python programing and the camera is recognize with the code. 

Not sure where to go from here. Any suggestions, please reply

Thanks

ccrisan, 2016-08-20:
Which image variant have you downloaded? Have you extracted it before writing it to the card?

pjockey, 2016-08-20:
I've figure this, was trying to put the RP 3 image into a 2. Fixed,my bad. I have the system booted up and connected to my network with wifi. It stops when it asked for the user ID. I entered admin with no password and I'm sitting at the CLI. Now what do I do?Scott                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 smartphone on the TELUS network.                                                                                                                                                                                                                From: Calin CrisanSent: Saturday, August 20, 2016 16:05To: ccrisan/motioneyeosReply To: ccrisan/motioneyeosCc: S Milliken; CommentSubject: Re: [ccrisan/motioneyeos] Raspberry PI Zero (#182)Which image variant have you downloaded? Have you extracted it before writing it to the card?

ccrisan, 2016-08-21:
@pjockey have your read [this faq entry](https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-am-only-getting-white-on-black-text-on-my-monitor-where-is-my-ui)?

P.S. why have you chosen to comment on this particular issue? You problem is not even remotely related to Raspberry PI Zero, nor the WiFi.

TheKitro, 2016-08-26:
Hi 
I cant geht my wifi config work :(

Situation:
Windows 10   #Workpc
Notepad++   #Editor
Raspi Zero 1.3 
Connected:
Night Vision Camera
HDMI
USB Hub  (Wifi Dongle,  Mouse - Keyboad Logitech Dongle)
SD Card 8 GB

What I done under Windows:
Download motioneyeos-raspberrypi-20160801.img.gz
Extract motioneyeos-raspberrypi-20160801.img
Format SD with SDFormater
Write Image to SD with Win32DiskImager
Create  wpa_supplicant.conf file on SD with Notepad++
Insert Text:
"update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
    ssid="your_network"
    psk="your_password"
}"
Replace your_network with my networkname  (e.g.  ssid="myprivateLAN")
Replace same way the psk
Change encoding of file in UTF8  and  UNIX Linebreak.
Save file
Plug SD in Raspi Zero
Boot
Get reboot circle. Error no wired network: no device  :(((

What do i wrong ?

Best Regards  
Cornflake
PS: Very nice tool I used it on good working in Raspi2 too.

ccrisan, 2016-08-27:
Make sure to:

1. name your `wpa_supplicant.conf` file correctly.
2. place your `wpa_supplicant.conf` in the correct directory (the root of the boot partition, next to e.g. config.txt)
3. place your `wpa_supplicant.conf` on the boot partition *before* booting your OS from the SD card for the first time.

P.S. if there's no mention of WiFI or wpa_supplicant in your boot log, then the file is not taken into consideration and most likely you did not follow one of the above conditions.

TheKitro, 2016-08-27:
It WORKS :)))
Ok my fault was to boot it first time without the wpa_supplicant.conf file. in the second boot this file wasn't included.

Additional Question:
I'd like to use this zero motioneye  on my friends home.  For this I have later to change the WLAN config. Is it possible to add 2 WLAN configs in the wpa_supplicant.conf file?  or change this file later over windows textfiles?

Big thx for your response and help
Cornflake

ccrisan, 2016-08-29:
motionEyeOS does not support more than one WiFi connection. If you must use two WiFi connections, please try Raspbian + motionEye.

calyerm, 2016-11-03:
Pi Zero:
Using 20160915 + usb network dongle ... this is working ... but
Limited menu options (Example no Motion control option ) ... 
Very low frame rate .. 2 fps ...  using  standard OS and other streaming methods , frame rates for 640x480 ..are good ..

fideocam, 2017-03-23:
Has anybody tried the new Zero W with motioneyes? I guess you could configure wifi password into the img before burning and just check the IP from wifi AP.

wdawson001, 2017-03-23:
@fideocam Follow the instructions here #804 (comment)

The official releases haven't been updated but I recompiled from the latest repo, so it works.

kiridashi, 2017-04-05:
@fideocam

running this configuration since a week (zeroW+NoirCam+motioneyeos)

setup was easy, just created a "wpa_supplicant. conf" file an entered wifi- ssid and -key.