motioneye-project/motioneyeos #183: Is it too soon to ask... Pi 3 Compatibility?

mehtaman, 2016-02-29:
Hi ccrisa,

I was just wondering if you would be looking at adding the various new drivers/packages into MotionEye OS for the new Pi 3 board?

ccrisan, 2016-02-29:
Wait till I get my hands on a PI3.

Mikiya83, 2016-03-01:
And please don't forget the Odroid C2 after the C1 that you already support, i find it more interesting than RPi3 for the same price, if it's supported by motioneyeos, it could be perfect ! :D

samg2016, 2016-03-02:
Added motioneye to my new pi3 yesterday. Runs very well.

db414813, 2016-03-03:
Yes. Odroid c2 support would be fantastic! Don't want to overwhelm you with boards though. Next thing ya know someone will be asking or orange pi support haha.

ccrisan, 2016-03-03:
Orange PI has already been requested a couple of times. The only real problem with many boards is the testing procedure right before a release, which is already painful.

db414813, 2016-03-03:
I wouldn't touch an orange pi. Horrible reviews on Amazon. I'll buy either a rpi3 or odroid c2, if it is supported. It would be my main server for my Motioneye set up. The c2 and the gigabit Ethernet is what has my attention.

ccrisan, 2016-03-03:
I hear you :)

alipepic, 2016-03-21:
hello,

we downloaded the pi 3 image of motioneyeOS. Having issues, in short administrative panel only has one new section, all the other sections of the administrative panel are gone. The frame rate is extremely slow . Advise please. thank you very much.

andrewvictor, 2017-02-24:
Having great success running several RPi3 integrated into one hub, on WiFi and uploading images directly to Google Drive.  Tested this platform throughout the winter (Canada, -35C) with great success!

bortek, 2017-03-01:
Great, I will therefore close this one.

closed 2017-03-01 by bortek