motioneye-project/motioneyeos #184: NOOB  question: Turn off Raspberry PI 2 status LEDs

Fourmyler, 2016-02-29:
Hi, I've got MotionEyeOS working and I have turned the Raspberry PI camera red LED off.  However, I can't figure out how to turn the onboard status LEDs off, which are making my homemade case glow red, when I would like to to be hidden.  I found the following link but, as a NOOB, couldn't work out what to do:

https://blog.bartbania.com/raspberry_pi/disable-raspberry-pi-status-leds/

Thanks in advance for any help.

JimPanski, 2016-03-21:
Hi ccrisan,
I tried out to disable the red power led by using the command mentioned in this guide [http://stackoverflow.com/questions/19863723/turn-off-leds-of-raspberry-pi](url) via ssh, but obviously the folder /sys/class/leds/ is empty, so the 'echo 1' command can't be executed.

It would be so nice if you could implement the option to disable/enable the leds by toggle a switch like the one for the CSI led :)
Or is there a workaround by running a command?

Many thanks for all the effort you already spent on this project - brilliant job!!!

Ethocreeper, 2016-04-27:
useless feature first he will add rtsp support and other stuff more usfull

giac0m0, 2016-08-04:
Adding a web gui control to implement controls to turn down or off the leds for the Pi in  `/boot/config.txt `is a good idea.

http://raspberrypi.stackexchange.com/questions/40559/disable-leds-pi-zero?answertab=votes#tab-top

shadowfxd, 2016-08-14:
SSH into your motioneyeos Pi2. Then enter these commands:

`mount -o remount,rw /`
`mount -o remount,rw /boot`
`nano /boot/config.txt`

at the bottom of /boot/config.txt enter this:

# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off

# Disable the PWR LED.
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

CTRL + X, then Y, then enter, reboot the pi through motioneyeos

gorgonops, 2016-09-04:
@shadowfxd Thx - did c&p your lines into the config.txt. Lights are both still on. I am on 20160705 with PI2. What am I missing?

shadowfxd, 2016-09-04:
make sure none of the lines are indented and that they are all on the very bottom of /boot/config.txt.

gorgonops, 2016-09-05:
@shadowfxd : Thx a lot. config.txt looks as follows: 
start_file=start.elf
fixup_file=fixup.elf
arm_freq=700
core_freq=250
sdram_freq=400
over_voltage=0
gpu_mem=128
gpu_mem_256=128
gpu_mem_512=128
disable_camera_led=1
gpu_freq=250
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

What am I missing? Mind is blocked.

shadowfxd, 2016-09-05:
this is exactly what my /boot/config.txt looks like on pi2 motioneyeos 20160705:

gpu_mem=320
arm_freq=950
gpu_freq=250
sdram_freq=450
over_voltage=0
disable_camera_led=1

# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off

# Disable the PWR LED.
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

gorgonops, 2016-09-05:
C&P of your full config.txt worked fine. Somehow I was lost with my german keyboard. Many thanks for yr help.

shadowfxd, 2016-09-05:
no problem!

bortek, 2017-03-01:
I added this to Wiki.

https://github.com/ccrisan/motioneyeos/wiki/Turn-Off-LEDs

closed 2017-03-01 by bortek

pimuser, 2017-03-12:
I have tried this with my Raspberry Pi 3. But the LED's are still on. Here is my C&P from the config.txt:
gpu_mem=128
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
dtparam=audio=on
disable_camera_led=1

# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off

# Disable the PWR LED.
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

roycie1282, 2017-03-17:
pimuser the RPI3 uses the LEDs through the GPIO (well the power LED actually) there is a way to turn them off using Raspbian that @6by9 has shown on this issue here https://github.com/raspberrypi/linux/issues/1332 
I am currently trying to find a way around this in MotionEyeos but have yet to get it done. Hopefully my buddy is gonna come help me out soon and see if it's possible. If anyone has any ideas please let us know.

roycie1282, 2017-03-17:
Also Bortek can you please label the wiki that code is only good for RPI AND RPI2, but not the RPI3

6by9, 2017-03-17:
Please try updating via rpi-update to the latest 4.9 kernel (usual warnings of course).
I wasn't aware of those particular dtparams, but the power LED is now using the normal device tree nodes, so changes may well work. If not then please raise an issue on raspberrypi/linux and we can look into it. I don't believe there is anything fundamentally stopping this behaving in the same way as the other platforms now.

bob-lee, 2018-11-29:
Has anyone managed to turn off PWR led of old RPI model B (BCM2835 Revision 000e)?
Above instruction doesn't seem to control it at all, bright red PWR led is always turn on..

jasaw, 2018-11-29:
From memory, I don't think PWR LED can be turned off on the original RPi B.