motioneye-project/motioneyeos #186: starting wpa_supplicant: failed

EricFries, 2016-03-01:
I had everything set up fine using ethernet.  But when I added my wireless settings, I am now stuck in a  reboot loop.  The errors are:
starting wpa_supplicant: failed
configuring wireless network: no link

It then fails to set the current date stops motioneye (and everything else) and reboots.  Any ideas?

ccrisan, 2016-03-01:
1. What OS version are you running?
2. What WiFi adapter are you using?
3. What security does your WiFi network have? WPA? WEP? Open?
4. Do you have any special characters in your WiFi password?

EricFries, 2016-03-01:
1. The 20160221 version for Raspberry Pi 2.
2. Edimax Ew-7811Un
3. WPA2.
4. No special characters, just a combination of upper and lower case letters.

Thanks for the response @ccrisan .  I'm only noticing now that OS version is prerelease.  I'm also not sure whether the wi-fi adapter requires any additional setup or if any drivers are preinstalled.

ccrisan, 2016-03-01:
Could you please run the following commands (via putty/SSH) and see if the problem persists:

    mount -o remount,rw /
    mv /etc/modprobe.d/modprobe.conf /etc/modprobe.d/modprobe.conf.bak

Then reboot.

EricFries, 2016-03-02:
Worked beautifully.  Thanks!

ccrisan, 2016-03-02:
@EricFries could you please further help me identify the root cause of the problem? Could you please reenable the modprobe.conf file:

    mount -o remount,rw /
    mv /etc/modprobe.d/modprobe.conf.bak /etc/modprobe.d/modprobe.conf

Then edit the file:

    nano /etc/modprobe.d/modprobe.conf

and change `rtw_ips_mode=1` to `rtw_ips_mode=0` (then Ctrl-O, Enter, Ctrl-X).

I suspect the `rtw_ips_mode` parameter could be the issue. The rest of the parameters are welcomed as they disable power management and USB suspend.

EricFries, 2016-03-03:
Happy to help, but I'm having trouble using SSH to gain access to the Pi.  The password I use to login in to the web server isn't working to log me in.  Anything I should try?

ccrisan, 2016-03-03:
Actually you're the second one to report a login issue. I'll try to see if there's a bug indeed about this.

ccrisan, 2016-03-03:
I cannot seem to be able to reproduce the problem. Could you please insert your SD card into a (Linux-based) PC and try to recover the /etc/motion.conf file from the data partition? If that's not convenient for you, could you please prepare an OS image from your current SD card, compress it and send it over to me?

joejoe4, 2016-03-06:
I have same problem as EricFries. Please Help. Thank you!

ccrisan, 2016-03-07:
A new version is on the way. I'm hoping to release it tonight. Hold on for another couple of hours please.

ccrisan, 2016-03-07:
@joejoe4 just to be clear, you're running 20160221 and you're facing wifi startup problems, right?

EricFries, 2016-03-07:
@ccrisan I'm on OS X, can you point me in the direction of instructions to get you an image of the SD Card.  Tried using `dd` but it never seemed to finish.

ccrisan, 2016-03-07:
`dd` is fine. Just let it finish. It should take tenths of minutes, depending on your card's capacity. After that, a `gzip your_file.img` should reduce the file size to something around 50 MB.

joejoe4, 2016-03-07:
@ccrisan  answer your question about facing wifi startup problems. The answer is yes. Also I've noticed that my camera led is on, but trouble using SSH and login (putting Ip address) into web server is not working.

joejoe4, 2016-03-07:
@ccrisan  forgot to add the comment. Sorry. I will go ahead to use your new version (20160306) and will remove 20160221 in SD. Many thanks!

EricFries, 2016-03-10:
@ccrisan Using gzip the image ends up being 408MB.  The original SD card was 32gb.  Is there an easy/realistic way for me to send this to you?

ccrisan, 2016-03-11:
Then you probably had some media files on the disk (pictures, movies). I don't mind downloading 400+ MB of data if you upload it to Dropbox or Google drive. Just make sure to email me the download link directly instead of publishing it here, for privacy's sake.

Snippo, 2016-03-12:
I just had the same problem (wpa_supplicant: failed and boot loop) with multiple MotionEyeOS versions. In the end I changed my Wi-FI USB adapter for another Edimax Ew-7811Un and it started working again. The old adapter worked fine previously so I have no clue what the problem is, but it will help you find the issue.

ccrisan, 2016-03-15:
@EricFries I booted your image on a RPI2 and everything seems to be alright (from the SSH/remote login point of view). What is the problem that you're having these days?

EricFries, 2016-03-15:
It was an issue loggin in with SSH.  You had asked me to try and help you figure out something that went wrong with the wpa_supplicant: fail error I was getting.  But I couldn't login with SSH.  The password I use to login through the webserver wasn't working for SSH.  Was I using the wrong password?

pgkool, 2016-03-16:
I am using the 20160306 build on a RPi1 and am having similar issues. The wpa_supplicant fails and goes into a reboot loop.

I tried moving the modprobe.cfg file as suggested above, but same issue.

I am also using a EdiMax wifi dongle...model Ew-7811Un.

I do have hidden SSID, no special characters in password, and I'm using WPA2.

pgkool, 2016-03-16:
@EricFries If I remove the wifi dongle, I am able to stop the reboot loop and ssh into the Pi.

ccrisan, 2016-03-17:
@pgkool make sure your wifi settings are correct and that no special characters are involved. Also make sure your power supply is good enough and can handle all the current drawn. You could try to remove all USB devices (if any) except for the wifi dongle and see if the problem persists.

pgkool, 2016-03-17:
@ccrisan I verifed there are no special characters and have reinstalled the entire motioneyeOS to see it it percists. I went out and got 2 more power supplies rated at 5V/2.0 amps, and even a different ralink based wifi dongle.  Same issue. Are there any log files I can look at?

pgkool, 2016-03-17:
@ccrisan Ok well i fixed it. Im not sure what really did 100% but I switched out alot of things. I tried a 2A and 2.1A power supply, 3 different wifi dongles and also changed the name of my wifi network, changed the security from WPA2-Personal(AES) to WPA2-Auto-Persoanal (AES + TKIP) and unhid the SSID.

It finally worked! 

Then I undid all this, went back to my original 1A power supply, original EdiMax Ew-7811Un wifi dongle, changed the WiFi network name back aswell as back to WPA2-Personal(AES), and hid the SSID again, and it still works.

I think the trick was to unhide the SSID...let it pick up the network...something stuck...and then after "priming the pump" i was able to hide it again and it still works.

rich02035, 2016-05-17:
I was having the same problem, then I noticed I was entered the ssid all lower case, when I entered the case sensitive ssid, the problem resolved.

AdamMalkin, 2016-06-06:
I just had this problem and fixed it based on rich02035's comment above, i had put a capital at the start of the "Wireless Network Name" where there should not have been one

cpugeek75, 2016-08-24:
I'm stuck in the infinite loop situation.  My EdiMax wifi dongle went out so I can't connect to it.  I am plugged directly into my router and still get this.  Hope to have new wifi dongle today or tomorrow but would like to get this resolved before then.  Thanks!

ccrisan, 2016-08-24:
@cpugeek75 have your read [this](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems)?

bortek, 2017-03-01:
Closing this assuming the problem is solved. Otherwise reopen.

closed 2017-03-01 by bortek

LimeBlast, 2017-11-27:
I'm having this exact issue on a Pi Zero W. It was working fine up until the point that I changed the admin password (via the web interface), then simply didn't manage to reconnect after.

I'm not able to plug in a network adaptor, nor use a different wifi adaptor becuase it's built in.

Not sure what to try next.