motioneye-project/motioneyeos #188: [RPi 2, Simple MJPEG] empty section "File Storage"

argenistobago, 2016-03-02:
Hi,

I've got motionEyeOs on RPi2 (latest release).
Successfully added Simple MJPEG Camera (Dahua). I can see stream, video is ok.

Unfortunately in section "File Storage" I have only one position "Run A Command".
I want to record on USB HDD.
There were no more changes in any config files.

Is it a bug?
Can I set the path in different way?

[image]

ccrisan, 2016-03-02:
Don't use simple mjpeg camera if you want motion detection and other options.

closed 2016-03-02 by ccrisan

argenistobago, 2016-03-02:
I want only save video on HDD. Is it impossible for MJPEG?
I cannot use RSTP because I receive unresolved error (rtsp netcam returned erroneous response: RTSP/1.0 401 similar to [)](https://bitbucket.org/ccrisan/motioneye/issues/198/d-link-dcs-935l-rtsp-error)

ccrisan, 2016-03-02:
Use Network Camera then.

argenistobago, 2016-03-02:
This answer resolves problem :) Thanks a lot.