motioneye-project/motioneyeos #190: Restore configuration failed

Priso19, 2016-03-03:
Hi, after putting some accents (é, à, è) in a Prowl web URL Hook  my eyemotion server crash:
it says "recommended refresh"

After the refresh, my four camera disappeared (they were not "Fast camera").

So I wanted to backup my configuration on a new image disk, message from motion eye : 
"Failed to restore the configuration!"

What have I doing wrong ? 
Sorry but I have no log file because I reset my SD card...

Config :
_Mac OS 10.7.5 on a MacBook dual USB
Safari 6.1.6
motioneye on a ethernet Raspberry Pi 2 with no camera on it
Current Version : 20160221_

Here my motioneye-config.tar file :
[motioneye-config.zip](https://github.com/ccrisan/motioneyeos/files/157544/motioneye-config.zip)

ccrisan, 2016-03-04:
Please never attach files that contain sensitive data. I have removed your attachment now, I have it on my laptop and I'll try to reproduce the problem.

pumbaa2183, 2016-03-05:
I have had a similar issue after trying to restore a backup
motioneye on a wifi connection Raspberry Pi 2 with one camera
Current Version : 20160221
MacBook Pro  OSX 10.11.2
Safari 9.1

ccrisan, 2016-03-06:
@Priso19 I cannot seem to reproduce your problem. However the file you have attached (which I have removed due to sensitive data) is a .zip archive that contains a "motioneye-config" folder and __MACOSX folder. It doesn't look like a configuration backup file created by motionEyeOS (which is a .tar.gz file with all the files in the root of the archive).

ccrisan, 2016-03-06:
@pumbaa2183 what are the exact steps that you follow when you get this error? Please walk me through it.

Priso19, 2016-03-06:
First, thank you for your attention about my sensitive data.

 Indeed I had opened the archive file to view its contents, sorry.

For me, on MAc OS 10.7.5 (Lion) and Safari 6.1.6, the backup button generate a .tar file not .tar.gz
Perhaps Mac OS X automatically decompresses the file after the download?

Priso19, 2016-03-06:
I found the backup file .tar.gz in the trash !
Renamed more with a number incremented according to the number of existing file :
"motioneye-config.tar 10-23-45-020.gz"

I took it out of the trash and then remove the unnecessary character and it works correctly.

[image]

ccrisan, 2016-03-06:
Alright, I'll close the issue then.

closed 2016-03-06 by ccrisan

Priso19, 2016-03-06:
The simple solution would be to request a name for the backup file and check if there is already one with the same name on the disk to delete it before saving.

Because Mac OX anyway unpack the archive (first level) and renames himself the existing file without asking his replacement :

"motioneye-config.tar"
"motioneye-config-1.tar"
"motioneye-config-2.tar"

Thank you for your time Calin.

pumbaa2183, 2016-03-06:
Clicked backup in the web UI and saved a back up file — for some reason its saving it as an .img.gz and not a tar.gz ( see screen 
shot) 

I am running the latest version of Motioned OS using MacBook Pro 10.11.4 and safari 9.1

ccrisan, 2016-03-06:
Please attach the screenshot.

pumbaa2183, 2016-03-06:
sorry about that -- apparently the screen shot I sent from email didn't post :( 
Update : I did just create a new back up by clicking back up and it is saving it as a config.tar file 
[image]

[image]

Priso19, 2016-03-06:
What is the size of the img file? Are you sure it's not the motioneyeos archive? Not the backup file ?

pumbaa2183, 2016-03-06:
My Bad the .img.gz is 230mb that must be the archive the .tar is 41 kb

Chiny91, 2017-05-30:
I know this is an old issue but it could do with a tidy conclusion; it would have helped me anyway 😄    For those using macOS/Safari, Safari (rather than macOS) by default partly unpacks the backup archive making it unusable to motionEyeOS.

A workaround is to change the Safari default:
Safari->Preferences->General 
Uncheck _Open "safe" files after downloading_

jordipalet, 2019-06-07:
Hi,

I did a backup of my configuration (5 cameras), and saved it correctly.

Upgraded with pip install motioneye --upgrade and the cameras are gone.

I restored the backup and restarted and still say no cameras defined ...

even if I restore manually the contents of /etc/motioneye, it looks like not recognizing the config files.

Any hints?

Numline1, 2021-12-30:
Just to add to this, if you end up in a situation where Safari decides to unpack the file, leaving you confused with your `.tar` archive in `Downloads` folder, the original file (`.tar.gz`) will be in your Trash, so you can just restore it and import it into MotionEye.