motioneye-project/motioneyeos #192: Noir Cam No Auto Exposure

peternsomerset, 2016-03-03:
Hi,

I install motioneyeos OK but when I tick the box for auto,  the screen cycles through about 6 levels of brightness.
My pi noir works OK with raspian and motionos works with microsoft USB cam.
Any suggestion as to what is causing this to happen?
Thanks

ccrisan, 2016-03-04:
Don't enable auto brightness (if that's what you mean by the "box for auto") for RPi cameras.

peternsomerset, 2016-03-04:
OK, is that the same for motion as well or just motioneyeOS that does not have this function?
Thanks
--
Sent from myMail app for Android Friday,

ccrisan, 2016-03-04:
It's motion's internal function. motionEye(OS) doesn't interfere in any way with the video processing.

peternsomerset, 2016-03-04:
OK thanks.

closed 2016-03-04 by peternsomerset