motioneye-project/motioneyeos #193: Carl9170 not found

yunusar8888, 2016-03-04:
My wifi dongle does not work. It should using Carl9170 firmware, but the firmware not found. I try apt-get install firmware-linux-free, not work.  

[  466.289268] cfg80211: Calling CRDA to update world regulatory domain
[  466.345184] usbcore: registered new interface driver carl9170
[  466.371515] cfg80211: World regulatory domain updated:
[  466.371537] cfg80211:  DFS Master region: unset
[  466.371545] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  466.371558] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  466.371569] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  466.371579] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[  466.371591] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  466.371603] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  466.371614] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[  466.371624] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[  466.371634] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[  488.625810] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
[  488.756184] usb 1-1.2: New USB device found, idVendor=1b75, idProduct=9170
[  488.756200] usb 1-1.2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[  488.756211] usb 1-1.2: SerialNumber: 12345
[  488.829810] usb 1-1.2: reset high-speed USB device number 4 using dwc_otg
[  488.959028] usb 1-1.2: Direct firmware load for carl9170-1.fw failed with error -2
[  488.959053] usb 1-1.2: firmware not found.
'

ccrisan, 2016-03-04:
I'll make sure to add the firmware to the OS image. I won't be able to test it though.

ccrisan, 2016-03-06:
The firmware has been added to the image.

closed 2016-03-06 by ccrisan