motioneye-project/motioneyeos #194: Dynamic DNS

physast, 2016-03-05:
How can you use dynamic DNS service like No IP?

Gordon-1, 2016-03-07:
Perhaps by following the [directions ](http://www.noip.com/support/knowledgebase/getting-started-with-no-ip-com/)at the No-IP web site.  If you need more help than that, I would suggest that you go to other support forums since setting up DDNS is not related to any issue with the MotionEyeOS software and I doubt that maintainers of MotionEyeOS have time to deal with unrelated issues.

Banditen01, 2016-03-07:
https://github.com/ccrisan/motioneyeos/issues/22

ccrisan, 2016-03-07:
More and more users request this DDNS functionality to be integrated with motionEyeOS. I see their point but this is normally something that's beyond the scope of a camera firmware/OS. It's about publishing your public IP address (if any) to a public (D)DNS service. It's something totally unrelated to the camera or the OS behind it. It's something that any other computer (or router) on the LAN could do.

However given the large number of such requests, I'll consider adding a few of the most common DDNS providers to motionEyeOS, configurable via the UI. I'm not promising anything and I'm not capable of coming up with an ETA at this moment.

Gordon-1, 2016-03-07:
I did not read this as a feature request and IMHO you should not be reinventing the wheel.  Still, you have to satisfy your customers.

@physast .. I would try to help you if you have problems setting up DDNS but you have to do your homework and research first.  If you need help after trying to set up DDNS then maybe add your email address to your profile since it is "beyond the scope of a camera firmware/OS."

By the way, I have run servers at home a lot using different DDNS services, but I prefer to use a VPN.  In other words, keep everything (including MotionEyeOS) on your local LAN and then use VPN  (and a DDNS) to access the LAN. I think this is more secure and less tempting to hackers than having open port(s) to server(s) protected only by a password. Some routers have DDNS and/or VPN built in (such as my Asus RT-N66U)

physast, 2016-03-08:
Thank you for the suggestions

closed 2016-03-08 by physast

aubvmk, 2016-12-03:
Hello everyone,
I have make a fork with include inadyn package and add a boot startup.
For the config add inadyn.conf in /boot

Here: https://github.com/ViaMonkey/motioneyeos

Auban

Roby86, 2016-12-09:
Great, thanks for your commitment, that was exactly what I was looking for. :-)
How can I flash your motioneyeOS on sd card? 

Can you build tar.gz image on the same way ccrisan does (tar.gz images)? 

https://github.com/ccrisan/motioneyeos/releases

Thanks! :-)

aubvmk, 2016-12-11:
Hello,

there is a build tar.gz in release for raspberrypi3.
For flash your sd card extract the image and write on your sd card with this tool for windows.

https://sourceforge.net/projects/win32diskimager/files/latest/download

enjoy

Auban