motioneye-project/motioneyeos #197: Motion detection outside set working scheedule for detection.

Banditen01, 2016-03-06:
One of two cameras are sending motion detection notice (without pictures attached and no recording are made) still sending emty detection notice to my e-mail. Any clue what's wrong?

ccrisan, 2016-03-07:
What's the connection between the title and the description of the issue? Do you expect pictures attached to emails or do you not expect any email at all? Please attach your configuration (removing sensitive data from it).

Banditen01, 2016-03-07:
Sorry, I understand your confusion, one of my cameras are sending e-mail motion detection notice when it´s supposed to be in non motion detect mode, set up by working Schedule function. It saves picture/movies on local disk but do not attache any pictures in e-mail and do not upload to Google Drive (even if I get okay from test service bottom). After some motion detetect on and off and also Schedule on and off and a reboot it sems to work again.When you ask for configuration, do you want all five different log files?

ccrisan, 2016-03-07:
I need the thread-*.conf and motion.conf files that are in the archive that you get when you click the Backup configuration button. Always remember to remove sensitive data from them.

quantum8, 2016-07-07:
I am getting a similar issue where it is recording motion events when they happen, regardless of being inside or outside the schedule. I'm running 20160605 on a rpi model b.
[Config.zip](https://github.com/ccrisan/motioneyeos/files/351152/Config.zip)

ccrisan, 2016-07-07:
@quantum8 So what you're saying is that you get motion movies outside of the configured working schedule?

ccrisan, 2016-07-07:
Ah I just noticed your working schedule is invalid. The "from" must be less than the "to" field. You cannot have a working schedule for a day span to the next day.

quantum8, 2016-07-07:
That makes sense, I'll change the settings and see how it goes from there.

closed 2017-03-01 by bortek

geoffp13, 2019-07-04:
Why on earth !!! - :-( I give up - so frustrated with this software - it is so close to working yet fails through simple errors in construction and explanation. Needs a little common sense and forethought !!