motioneye-project/motioneyeos #198: Raspberry Pi 2 Model B - Increasing GPU memory causes system to fail.

Triavati, 2016-03-07:
Everything works fine when I have my GPU memory set to 128 and the overclock speed at "Pi2". When I increase the GPU memory to 256, and overlock to Pi2, I get the "Reload recommended" error, and am forced to reinstall the MotionEyeOS to get it working again. 

It's my understanding that I dictate the GPU memory, and what is left over goes to the CPU. So if the pi 2 has 1024 MB of available memory, assigning 256 to GPU should leave plenty left over for system processes. Am I missing something?

ccrisan, 2016-03-07:
I'll look into it.

Triavati, 2016-03-07:
Thank you! I'm a true novice in the raspberry pi community, so I'm 90% sure that my problem is user error. MotionEyeOS is amazing, and I'm currently using the motion detection to protect my valuables from my roommate's thieving friends.

I'm trying to optimize my settings so that I reduce the frame-rate drop I encounter when motion is detected. It's trying to stream, record, and detect motion at the same time (640 15fps), which is what lead me to adjust the memory and clock speed. Any advice would be greatly appreciated! Keep up the good work.

Priso19, 2016-03-11:
Same on RPi3

ccrisan, 2016-03-15:
The bug has been fixed and the fix will be included with the next release.

closed 2016-03-15 by ccrisan