motioneye-project/motioneyeos #199: Disabling camera LED on Raspi3 gives error

tampano, 2016-03-08:
Clicking apply after disabling the camera led asks if it's ok the system will reboot. Clicking OK gives an error and the system doesn't reboot. After manual reboot camera LED is still on despite the choice.

Gordon-1, 2016-03-10:
Behavior is confirmed (RasPi3 and RasPi camera).  Setting camera LED to off then “apply”, screen shows “this will reboot…”   Choose yes to reboot then screen shows “An error has occurred, refreshing is recommended”.  System does not reboot and after reloading the web page “apply” button still shows.

motioneye.log entry at the time the above actions are taken:

2016-03-10 07:26:29: [motioneye]    ERROR: global name 'arm_freq' is not defined
Traceback (most recent call last):
  File "home/ccrisan/motioneyeos/output/raspberrypi3/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1401, in _stack_context_handle_exception
  File "home/ccrisan/motioneyeos/output/raspberrypi3/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1603, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 229, in post
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 158, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 451, in set_config
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 357, in set_main_config
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 139, in set_main
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 1911, in _set_additional_config
  File "/usr/lib/python2.7/site-packages/motioneye/boardctl.py", line 143, in _set_board_settings
NameError: global name 'arm_freq' is not defined
2016-03-10 07:26:29: [motioneye]    ERROR: 500 POST /config/0/set/?_=1457612773035&_username=admin&_signature=73bd8530811c883ad0bbef8f4aa2825ac5b32d1b (10.178.34.22) 28.86ms

:

Priso19, 2016-03-11:
+ 1

ccrisan, 2016-03-15:
The bug has been fixed and the fix will be included with the next release.

closed 2016-03-15 by ccrisan

Priso19, 2016-03-19:
Until the update is available I edit the config.txt file in the root of the SD card with an SD card reader. This seems to work.

gitunit, 2017-07-20:
still doesn't work toggling the camera LED. what params did you add @Priso19 ?