motioneye-project/motioneyeos #202: Upgrading from v20160109 on Pi B+ via webui

sinbrkatetete, 2016-03-10:
Hi!

I've got two Pis running MotionEyeOS. One quite near me (behind my back) which is used primarily as a fire alarm of sorts (Pi 2 with Pi cam) and one about 300 km away which is used as a security cam (Pi B+ with PiNOIR cam). The one right behind me did both of the updates since the version in the title without a problem via "check" and "update" from webui, but the other (older) one, the B+ claims the version installed (v20160109) is the latest available version and, consequentially, doesn't offer the update option.

Is there anything I can do to make the distant one "see" the available updates and update it via webui, or do I have to wait 'till I go to the distant location and do the update manually? 

Thanks in advance.

ccrisan, 2016-03-11:
20160109 is the latest *stable* version. Newer versions are prereleases. If you really need these prereleases, enable them in Expert Settings. But don't upgrade unless you really need features and fixes included with them.

closed 2016-03-11 by ccrisan

sinbrkatetete, 2016-03-14:
whoops. sorry for opening an issue for such a stupid miss on my side. thanks