motioneye-project/motioneyeos #203: Toggling motion detection on any camera causes lost connection to USB cam (camera 2)

Gordon-1, 2016-03-10:
RaspPi3 pre-release version 20160306, using RasPi3, RasPi cam and USB camera.

With power off plugged in Logitech C210 USB camera (which worked ok with RasPi2 and MEOS), booted up, added camera.  All seemed OK.  Turned off motion detection and applied,  camera 2 (USB) disconnected.  motion.log showed these lines, repeated ad-infinitum:

2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Input/output error
[2] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2] [ERR] [ALL] motion_init: Error capturing first image
[2] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
(log entries are also made in /var/log/messages)

Rebooted and all seemed well again (no errors in logs except the ever-present “Device doesn't support VIDIOC_G_STD”)

However, turned on motion detection for camera 1 and lost connection to camera 2 again with same repeating log error messages.

Powered off and relocated camera 2 to a different USB port, booted OS and tried turning motion detection on for cam 1.. that worked.  Then turned motion detection on for cam 2 and lost the connection again (with same errors in logs).

The same cam seems to work fine on RasPi 2 with MEOS but I have to wonder if the camera is the problem.  It would be great if someone else could try to duplicate this with a RasPi camera AND a USB camera.

I think we can rule out power supply issues.  I am using the 2.4 amp PS with no other USB devices other than the cam.

ccrisan, 2016-03-11:
Might this be related to the power supply? If you use the same PSU that you used with RPI2, the RPI3 is said to draw a bit more current than its predecessor.

Gordon-1, 2016-03-11:
Thats a good thought, and they is why I specially stated "I think we can rule out power supply issues. I am using the 2.4 amp PS with no other USB devices other than the cam." (2ed to last paragraph)
That is the PS meant for the RasPi3.  Unless it is defective I doubt that is the issue.

I still would not be surprised if my camera was flaky so if no one else can duplicate this then I would just suggest this be closed. I should mention again however that in the limited time I used this USB cam on a RasPi2, I never saw this issue.

Gordon-1, 2016-03-15:
I'm still seeing this issue on a RasPi 3 with RasPi Cam (#1), and with a different USB cam (camera #2)  than in my original report.  Specially, I turned ON motion detection on cam 1, applied, then turned it OFF and applied.  At that point the connection to the USB cam  (# 2) was lost and errors were reported in the logs (attached). Cam does not work until I reboot. Cam is the only USB device. Using a 2.4 amp PS.

[Issue203-Logs.zip](https://github.com/ccrisan/motioneyeos/files/173088/Issue203-Logs.zip)

ccrisan, 2016-03-15:
Well there's the following line in your log motion.log file:

    motion_loop: Video device fatal error - Closing video device

It's probably something related to a buggy v4l driver or some issue that has nothing to do with motionEye(OS).

closed 2017-03-01 by bortek