motioneye-project/motioneyeos #204: Missing Email Settings in GUI

sherm8n, 2016-03-10:
I set up to send email notifications the other day, when now when I login those settings are now gone. The whole menu section where to set it up is just missing. 

This is the menu closed:
[image]

Menu open:
[image]
[image]
[image]
[image]
[image]

I'm using a RPi 3 with MotionEyeOS.

sherm8n, 2016-03-10:
Okay, it seems that the "Fast Network Camera" option diables notifications. However, when try to disable that option, I get an error and am unable to disable it.
[image]
[image]

ccrisan, 2016-03-11:
Indeed, the Fast Network Camera mode disables all those motion-related options. The error you get is a known RPI3-related problem and will be fixed soon.

ccrisan, 2016-03-15:
The bug has been fixed and the fix will be included with the next release.

closed 2016-03-15 by ccrisan