motioneye-project/motioneyeos #206: Installinng dyndns client

darrenc90, 2016-03-12:
Hi Cailin,

Great bit of software here, my only issue going right now is that I don't know how to install dyn's dns client, or any other software for that matter, can you please walk me through it?

thanks,
Darren

Banditen01, 2016-03-12:
https://github.com/ccrisan/motioneyeos/issues/194

sinbrkatetete, 2016-03-14:
I've done it following the instructions on the [tweaks page](https://github.com/ccrisan/motioneyeos/wiki/Tweaks) Banditen01 linked and instructions on [duckdns](https://www.duckdns.org/), I use as the dynamic dns service (and recommend).

aubvmk, 2016-12-03:
Hello everyone,
I have make a fork with include inadyn package and add a boot startup.
For the config add inadyn.conf in /boot

Here: https://github.com/ViaMonkey/motioneyeos

Auban

closed 2017-03-01 by bortek