motioneye-project/motioneyeos #208: v20160109 RPi2 - Date Wrong (suspect NTP server not Running)

SteveScanlon, 2016-03-14:
I installed motion eye os v20160109 on a RPi2. The date defaults to 01/01/1970. I suspect that this is an NTP issues such as the server not running at boot.

ccrisan, 2016-03-14:
NTP is not used by default. Make sure your PI has access to the internet and that it can resolve domain names through a DNS server.

SteveScanlon, 2016-03-14:
Hello Calin,

Thank you for the quick response.

This is a strange one as I have two RPi's 2 running the latest version.
Both are using UTC and HTTP to get time.
One is OK the other just refuses to pick up the time.
Both are using the same image???

The faulty one does not appear to be able to resolve DNS names.
I SSH'd into the faulty PI.

Pinged google.com and bbc.co.uk and this failed.

[image: Inline images 1]

Any suggestions?

Steve

SteveScanlon, 2016-03-14:
Here is a screen shot with more detail.

[image: Inline images 1]

Kind regards,

Steve

ccrisan, 2016-03-14:
I'm not seeing any image. Have you set a static IP? What's the contents of the /etc/resolv.conf file?

SteveScanlon, 2016-03-14:
Yes I have a static IP address.

Will copy and paste images into a word doc as I don't think Gmail likes
images pasted directly into the message.

Please find word doc attached.

Kind regards,

Steve

SteveScanlon, 2016-03-14:
Hi Calin,

I do appreciate your time on this one, however I am sure you are very busy
its just very, very strange.

I will try to include a screen shot.
There should be an image below here....

[image: Inline images 1]

As you can see it thinks its 1970 which is the base date.

Kind regards,

Steve

h44p, 2016-03-14:
Same issue here. Date for some reason 01/01/1970.
[image]

larngold, 2016-03-14:
I've had the same issue. The problem seems to be that external routing isn't set up. If you log into the pi, enter the following at the # prompt:

route add default gw ip_address_of_your_router

then hit return

then run ntpdate [location].pool.ntp.org - so for instance, I ran ntpdate uk.pool.ntp.org because i'm in the uk. 

The settings aren't preserved on reboot - what file preserves them?

In my own case, wireless didn't work either (the device was seen, it just wouldn't get an ip via dhcp) - this was with 20160306. Maybe there's something amiss with what writeimage.sh does?

SteveScanlon, 2016-03-15:
You are correct.
The location for the ntp setting is /etc/ntp.conf

# nano /etc/ntp.conf

However even as root you do not have permissions to edit this file with
this build.

This is possibly a problem is you assign a static IP from the web gui.

ivr588, 2016-03-15:
you need to remount root (/) rw like this:

mount / -o remount rw

then edit the file, then save the file, then mount / -o remount ro

SteveScanlon, 2016-03-15:
OK the problem is when you configure a static IP address from the web gui
make sure that the Default Gateway (IP address of your router)
is correct.

Then after re-boot your Pi will have the correct address as it will be able
to resolve DNS names.

ivr588, 2016-03-15:
In the end, I re-did the microsd - this time without using writeimage.sh at all, installed the card, booted the pi and waited a few minutes to see a new dhcp lease/ip [ethernet] on the router then ssh'd to the new ip. Dunno what went on there- same image, hadn't re-downloaded it, just didn't use writeimage.sh.

h44p, 2016-03-16:
I didn't have time yet to do any configuration to my setup and just noticed today that date and time are correct now.

SteveScanlon, 2016-03-16:
I formatted the SD card and then wrote the 20160109 image again.
From the web gui when assigning a static IP make sure you have the
correct IP address of the Gateway (your router). This solves the issue.

closed 2016-07-02 by ccrisan

bradleyproject, 2020-01-21:
Hi 

guys i just got my google drive account linked up after a day and a half of tinkering, however i got it to work whilst on auto IP configuration is it possible to get it working when on static also. I noticed when i was using static IP configuration the data and time was incorrect it showed 1/1/1970. Is this connected to my issue, i cant seem to get the correct date and time when on static IP configuration.

many thanks in advance guys

kind regards 
David