motioneye-project/motioneyeos #209: Raspberry Pi ethernet musings

Chiny91, 2016-03-14:
Reading through various posts, I am beginning to wonder whether one of the causes of assorted R Pi problems (crashes, non-responsiveness etc) is related to the R Pi combined USB/ethernet hardware.   AIUI, the R Pi hardware has some kind of combined bus (100 Mb ?) for all USB & ethernet, which may be a bottleneck when using motionEyeOS as video streaming, multiple cams and uploading to the cloud, all at the same time.   I recall that @ccrisan has no problems with an Odroid XU4, which I believe has a Gigabit ethernet.   I find that occasionally, DropBox is up to 20 minutes behind in being updated - could be several issues here.

As an experiment, I recompiled motionEyeOS to include rsync, and dumped DropBox for a throttled (1 Mb/s) backup using _rsync --bwlimit_.   Video streaming remains the same at 1 frame per second.   My R Pi 2 is adequately coping with streaming, rsync and motion detection using 4 netcams with no crashes/freezing in 5 days so far, although motioneye still tends to reboot several times a day.

Just musing - not rigorous lab testing.

ccrisan, 2016-03-14:
Software-induced reboots leave some traces in /var/log/messages (usually containing the "rebooting" text). I'd like to know if you can spot any such messages to find out the cause of reboots.

Chiny91, 2016-03-14:
OK, chunk of /var/log/messages appended from a period of calm (08:38) through 2 reboot messages and back to normal running.

[messages.txt](https://github.com/ccrisan/motioneyeos/files/171948/messages.txt)

ccrisan, 2016-03-14:
@Chiny91 "motioneye: not responding for 120 seconds, rebooting" indicates that motion (and motionEye, in turn) is not responding for a considerable amount of time. From my experience, this often happens with motion when using a problematic USB camera, or sometimes with RTSP cameras.

Chiny91, 2016-03-14:
OK, I'm not using either.   These are IP cams (not USB) nor configured using RTSP.   The head-scratching continues.

digitizing, 2016-03-27:
What is the Frame rate set for under VIDEO DEVICE for each camera.

Chiny91, 2016-03-28:
In checking the R Pi, I find it appears to have died after 15 days.   I've just done a power cycle but it does not look like it will recover.   It does respond to _ping_ and _ssh_ (rejecting the password).

From memory, _the Frame rate set for under VIDEO DEVICE for each camera_ is set to minimum, that is 2.   All the cameras have the same settings.

closed 2017-03-01 by bortek