motioneye-project/motioneyeos #211: Canceled sign-in leads to loss of menu.

Gordon-1, 2016-03-15:
Starting with a valid user or admin logged in, if the user clicks on the “Switch User” icon (which I did by mistake) and then “cancel” on the sign-in pop-up, then subsequent attempts to display the menu only displays a grayed out partial menu until the user signs in again.  Perhaps this is by design as an alternative to a “Log-out” option.  See my screen grab.

Same occurs with RasPi 2 and 3.

[image]

ccrisan, 2016-03-18:
Well this is kind of a "logged out" state. Aren't you able to close the menu and click on the switch user icon again?

Gordon-1, 2016-03-23:
Yes  - After logging in again all is well.  I just did not expect to be logged out when canceling the inadvertently activated log-in screen and it was not intuitive that one would need to log in again.  

If this behavior is by design, or is not be design but still acceptable, then please move on to more important open issues.  THANKS

closed 2017-03-01 by bortek