motioneye-project/motioneyeos #212: Not a supported network camera

mrkenison, 2016-03-15:
I have a Foscam FI8910W that I am trying to connect to motioneyeos.  I get the error "not a supported network camera" after entering in the local ip address of the camera and the forwarded port along with the username and password.  Is there a list of wireless ip cameras that work with motioneyeos?  It is also possible that I messed something up on the setup but I am pretty sure I have it right.  

I appreciate any information you can give me.

Phesten, 2016-04-15:
I'm having the exact same issue with my FOSCAM FI9803EP camera. I am using a Raspberry Pi 3 board as my hub.

ccrisan, 2016-07-12:
Look for the MJPEG URL. You're probably entering the wrong URL.

goney3, 2016-07-12:
Hmm, looking through the manual it says the camera supports JPEG, MPEG4, H.264 ... I guess MJPEG isn't supported? It can also run in single codec or dual codec modes. My apologies if MJPEG is one of the others, I am very new to this. Thank you for your quick response, and from the creator no less! Thank you for the awesomness that is MotionEyeOS! :+1:

goney3, 2016-07-12:
I found a help-section on iSpyConnect that had a jpeg path creator for our Sony Cameras, it basically suggested:
http://user:password@IP.IP.IP.IP/jpeg/qvga.jpg

We got it working by making the path our HTTP://IP_ADDRESS/jpeg/qvga.jpg and entered our user/pass below like normal, then MotionEyeOS then recognized it as a JPEG Network Camera and now works beautifully! :)

We also tried vga.jpg for the larger size (as qvga is 320x240). I hope this helps others who don't know why their network camera isn't showing up as a supported network camera in MotionEyeOS! We looked through all 111 pages in the manual and couldn't find the path, so maybe a link over to the iSpyConnect finder would help other MotionEyeOS users.

https://www.ispyconnect.com/sources.aspx

closed 2017-03-01 by bortek