motioneye-project/motioneyeos #213: Showing cpu usage / temperature

molly78, 2016-03-15:
Much thanks for an awesome piece of software :)

Any chance you could add the cpu usage / Pi temperature on the main page? Top corner perhaps?
Maybe memory allocations as well?

Help to see if you're overloading the Pi

Playing with it on a new pi3 and trying to trouble shoot when two cams won't display. 

Again huge thx :)

funtomas, 2016-04-05:
for cpu temperature you can make your own script in "storage" directory

cd storage
nano cpu.sh
// add this lines in editor
#!/bin/bash
cpuTemp0=$(cat /sys/class/thermal/thermal_zone0/temp)
cpuTemp1=$(($cpuTemp0/1000))
cpuTemp2=$(($cpuTemp0/100))
cpuTempM=$(($cpuTemp2 % $cpuTemp1))
echo CPU temp"="$cpuTemp1"."$cpuTempM"'C"
//press CTRL+x for finish and SAVE file
chmod +x cpu.sh
// and run script
./cpu.sh

EpicLPer, 2016-05-23:
It would be nice to see this on the dashboard if possible, I'm getting weird errors with my Pi 1 and I'm not sure if it's in any way due to overheating and/or in a problem when the sun shines for a longer time on it.

bortek, 2017-03-01:
Not for now.

closed 2017-03-01 by bortek