motioneye-project/motioneyeos #215: Setting not available

uac988, 2016-03-15:
Just installed latest release 20160109 on Pi2, standard Pi Camera

Web interface is different to descriptions on web. No admin login key icon. Setting are limited to

Layout Columns	
Fit Frames Vertically
Frame Rate Dimmer	
Resolution Dimmer	

Literally cannot do anything with web interface. Real time monitor of camera does work in web browser

molly78, 2016-03-15:
Perhaps it looks like you're logged in as a user
Click on the user icon on the left to get the prompt to login

uac988, 2016-03-16:
Thanks for the fast feedback! I am away from the unit at the moment so
am running from memory.... I did click on the user icon. No prompt to
login appears. No much appeared to happen. Will double check again
tonight though.
Also it appears I could not login via SSH until I've logged in at
least once via keyboard. Is this normal?
David
----- Original Message -----
From: "ccrisan/motioneyeos" 
To:"ccrisan/motioneyeos" 
Cc:"uac988" 
Sent:Tue, 15 Mar 2016 07:53:48 -0700
Subject:Re: [motioneyeos] Setting not available (#215)

	Perhaps it looks like you're logged in as a user
 Click on the user icon on the left to get the prompt to login 

	—
https://github.com/ccrisan/motioneyeos/issues/215#issuecomment-196854863
[1]     

Links:
------
[1]
https://github.com/ccrisan/motioneyeos/issues/215#issuecomment-196854863

molly78, 2016-03-16:
did you reboot the device after the initial install boot up?

default login for admin is no password - should set that asap

Ethocreeper, 2016-04-27:
? for me it worked just fine it the pi1b+ and the pi3.
i just connected to them then trust there certificate the firs time and asks for login just enter admin in the username and hit enter. then i enabled advanced settings.

closed 2017-03-01 by bortek