motioneye-project/motioneyeos #216: Raspberry 3

duckietm, 2016-03-15:
Hello,

I installed the oh so nice Motion Eye Os on the raspberry 3 and all looks fine first hand, Thanks for this :) 
However a couple little points:

- if i make changes to the webinterface i sudenly recive a error: "An error occurred. Refreshing is recommended." and then nothing happens and making the change again same error, after a reboot i can change stuff again ? (to go straight to the error, use the disable light switch)

- Enabled the SSH, but it is not working ....

- Where are the overclock settings ?

ccrisan, 2016-03-15:
I'm working on the issue related to the errors you get. No overclocking settings are supported for the RPI3.

ccrisan, 2016-03-15:
The bug has been fixed and the fix will be included with the next release.

closed 2016-03-15 by ccrisan

duckietm, 2016-03-16:
Okie Thanks !

https://github.com/ccrisan/motioneyeos/issues/216#issuecomment-196875895