motioneye-project/motioneyeos #217: How to: Motion Notification delay

clawwd, 2016-03-15:
I have a camera on the front porch to monitor people coming and leaving.

My issue is that when the screen door opens, it triggers an event and I am sent a picture of the door opening rather than the person leaving.

Is there a way I can delay the picture that is sent so that instead of capturing the door I see a few seconds later?

clawwd, 2016-03-17:
Any ideas?

ccrisan, 2016-03-17:
Increase the Attached Pictures Time Span value?

clawwd, 2016-03-17:
I tried changing it to 10 seconds but the pics being sent via email doesnt show 10 seconds worth.

ccrisan, 2016-03-17:
Try to change the motion detection parameters then. It's never exactly 10 seconds worth of pictures. It's *at most* 10 seconds worth of pictures. The actual number of pictures taken during motion is decided by the motion daemon using the configured motion detection parameters, frame rate and so on.

closed 2016-03-17 by ccrisan