motioneye-project/motioneyeos #218: Messages file growing fast (66 MB)

dpatonsr, 2016-03-15:
The following lines are being repeated continuously for days at more than 1 per second:
Feb 29 18:59:33 meye-01ee10f5 user.warn motion-mrdave: [2] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
Feb 29 18:59:33 meye-01ee10f5 user.warn motion-mrdave: [3] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd

The cameras are also losing connection from my home wifi about once a week.  I have them setup with a static IP thru my Apple Airport extreme router.  When I force reboot the cameras they are available again. They keep working even though I can't contact them.  I would of course prefer not to have to keep force rebooting them!

Chiny91, 2016-03-17:
I've just noticed the same _messages_ lines myself; a single (of 4) camera provided that very warning for about 20 minutes before re-connecting successfully.    I'm using _motion-svn_ not _motion-mrdave_, although they do have common code, so that doesn't really say much.

As the problem is a _warning_ not _error_, I've solved the monster messages file problem by (re)setting:
_log_level quiet_in /data/etc/motioneye.conf
If you have the default _info_, then you will get an ever-increasing without limit, messages file.   @ccrisan is reviewing this aspect of motioneyeos.

bortek, 2017-03-01:
Havent been any updates since a year ago. Therefore closing.

closed 2017-03-01 by bortek