motioneye-project/motioneyeos #219: Installation

cbensen, 2016-03-16:
I'm running a Mac and I've had zero luck getting motioneyeos installed on an SD card. I have never had this issue with other Raspberry pi installations such as Raspbian. I've followed the directions and tried a few things not in the directions. writeimage.sh can't mount the file system, it wants ext4 file system support. I installed OSXFuse. OSXFuse doesn't support writing and unfortunately there isn't anything that's free that supports ext4 and has write support. So I tried VMware fusion with my Ubuntu VM. No matter what I try it can't see it. When I connect the USB SD card reader to the VM the only new files are sdb, sdb1 and sg2. This is where my knowledge of Linux falls short, I have no idea which one. So I tried all three and writeimage.sh didn't like any of them. So I'm posting here. How does someone with a Mac setup motioneyeos? And why does it require the ext4 file system rather than something that at least has one readable partition like Raspbian?

Thanks,
Chris

Chiny91, 2016-03-16:
Perhaps it would be easier to give the hard stuff (command line) and go GUI with [Apple Pi Baker](http://www.tweaking4all.com/hardware/raspberry-pi/macosx-apple-pi-baker/).   I find it is necessary to use the motioneyeos.img file (not the compressed version) but otherwise it is a matter of minutes to d/l, burn and be using motionEyeOS.

db414813, 2016-03-17:
Why don't you just use dd in terminal.

cbensen, 2016-03-17:
I haven't tried Apple Pi Baker, I'd give it a try but I don't run random programs on my computer that request my admin password.

dd said it worked, the pi doesn't boot. Just sits there with one red light, that's why I decided to follow the directions and use writeimage.sh.

ccrisan, 2016-03-17:
@cbensen motionEyeOS uses ext4 just as Raspbian does. From this point of view there's no big difference. The fact that OSX can't mount ext filesystems in 2016 is just too bad.

I planned however to use the fat partition for initial configuration some time in the future, so systems w/o ext support could access it. Until then you need to write the OS image using dd or some tool as mentioned by @Chiny91.

closed 2016-03-17 by ccrisan

Chiny91, 2016-03-17:
@cbensen There is an non-SUDO version of Apple PiBaker available if you don't want to provide an admin password.

cbensen, 2016-03-17:
@ccrisan, That's right, the reason I don't have a problem with Raspbian is because there's a FAT partition. That's why I switched to the writeimage.sh script.

dd works but I can't get the OS setup. Does anyone have any idea if I can do this from a Raspberry pi with Raspbian?

cbensen, 2016-03-17:
So I tried running writeimage.sh and dd from Raspbian. Why not, it's a Linux OS! writeimage.sh failed with an error similar to this https://bugs.launchpad.net/ubuntu/+source/parted/+bug/1220165

dd succeeded just like it does on my Mac, but the image doesn't boot. No activity nothing. Any ideas? How does everyone here setup motioneyeos? There's got to be a step I'm missing. Maybe I grabbed the wrong image, but I don't think so. I have a Raspberry Pi Model B so I made to I had  the image https://github.com/ccrisan/motioneyeos/releases/download/20160109/motioneyeos-raspberrypi-20160109.img.gz

The results are the same. The same Raspberry pi that I created the motioneyeos SD card image on doesn't have any ACT activity no matter what I do. How does everyone else setup this OS?

ccrisan, 2016-03-18:
@cbensen how do you know it doesn't work? Have you connected a monitor? Have you looked through your router's DHCP leases for meye-xxxxxxxx?

cbensen, 2016-03-18:
@ccrisan I know because the power LED on both the pi and the camera are red and the ACT and LCK LEDs do nothing, not a blink once I connect power. See attached photo. Yes I have connected a monitor, nothing, and yes I have looked at my router's list of DHCP leases and there's no new devices. I know the pi works because I have another SD card with Raspbian that displays and this last attempt to setup motioneyeos I used that Raspbian. So I know the pi works, I know the network and display work, I know the SD card reader works, I know this SD card works because I installed Raspbian for kicks just to prove it. The bootup sequence of LEDs does not match (just a static red LED vs a clicker of all LEDs). I can't mount the ext4 partition on the Raspbian either. It looks like the only thing that writeimage.sh does is setup the wireless network connection or to use a static IP. Otherwise it should boot just using DHCP and the built in wired ethernet, correct?

So what am I doing wrong? So how does everyone else setup this OS? Am I the only one doing it from a Mac, VM on Mac or Raspbian? I'd really like to get this working and contribute if I can :)
[image]

ccrisan, 2016-03-18:
Indeed, the only thing that writeimage.sh would do in addition to the traditional dd or any other disk image tool is to preconfigure your network settings.

Is there anything that shows up on the monitor? If you plug in your sdcard to your mac and type `fdisk -l` what does it say? (Do you have the `fdisk` command on mac?)

cbensen, 2016-03-18:
>fdisk -l
fdisk: illegal option -- l
usage: fdisk [-ieu] [-f mbrboot] [-c cyl -h head -s sect] [-S size] [-r] [-a style] disk
	-i: initialize disk with new MBR
	-u: update MBR code, preserve partition table
	-e: edit MBRs on disk interactively
	-f: specify non-standard MBR template
	-chs: specify disk geometry
	-S: specify disk size
	-r: read partition specs from stdin (implies -i)
	-a: auto-partition with the given style
	-d: dump partition table
	-y: don't ask any questions
	-t: test if disk is partitioned
`disk' is of the form /dev/rdisk0.
auto-partition styles:
  boothfs     8Mb boot plus HFS+ root partition (default)
  hfs         Entire disk as one HFS+ partition
  dos         Entire disk as one DOS partition
  raid        Entire disk as one 0xAC partition

cbensen, 2016-03-18:
I think I see where you're going, and I agree, the SD card must not be partitioned correctly. However, dd succeeds, so I'm at a loss.

ccrisan, 2016-03-18:
Damn these macs. So what do you see on your monitor? Does wake from sleep? Do you see the rainbow?

cbensen, 2016-03-18:
I see nothing on the monitor from the pi. There's no signal. Pop in the Raspbian SD card, turn on power it boots normal. I'm not sure it's the Mac, I ran dd from Raspbian too.

OK, so I know the pi and all the hardware is fine. So it must be something with the partitioning or copying of the image. dd succeeds so I'm assuming that's fine. I can't mount ext4 to verify so that's my biggest hurtle. Although I did have the ext4 read only driver installed at one point but writeimage.sh said I needed read/write so I uninstalled it. I can go back, later tonight, and install that. I could then verify that dd worked beyond it saying it succeeded and not reporting any errors. The only other thing I can think of is the partitioning. How do you need the SD card partitioned? How do you partition it? I'm assuming you use Linux since you use fdisk? So what would your fdisk command be to partition the SD card before you run dd or writeimage.sh?

ccrisan, 2016-03-18:
`writeimage.sh` internally uses `dd` and the partition layout comes with the image. Even if you would partition it using `fdisk` and then format the partition(s), the partitions would be overwritten by those found in the image file.

The partition layout used by motionEyeOS is the same as that used by Raspbian: one fat partition and one ext4 partition (at least at the beginning).

How does your `dd` command look like?

cbensen, 2016-03-18:
That's what I thought. My dd command is:

sudo dd if=motioneyeos.img of=/dev/sda1

Am I missing anything? I don't see a fat partition.

ccrisan, 2016-03-18:
Why are you using `/dev/sda1`? Shouldn't it be `/dev/sda` (i.e. the entire card, not just the first partition)? Are you sure your card device is `/dev/sda`?

cbensen, 2016-03-18:
Got it working! I'm on a different Mac right now so gave it a try there:

dd if=motioneyeos.img of=/dev/disk2 bs=1048576

The /dev/sda1 was on Raspbian, and that's where my knowledge of Linux gets a bit fuzzy. Yes I know that SD card was associated with sda or sda1. I tried both with writeimage.sh, both failed, I only did sda1 with dd. So, I can write up some simple instructions for Mac fairly easy if you're interested.

Thanks for all the help! Now I can dig into it!

cbensen, 2016-03-18:
@ccrisan So as I said I got it installed. I started looking at it and comparing to the screenshots (https://github.com/ccrisan/motioneyeos/wiki/Screenshots). My settings (attached) don't show all the settings. I've tried with Safari and Chrome on Mac and on mobile. Mobile has even fewer settings. Any ideas?

[image]

cbensen, 2016-03-18:
Never mind, admin with no password gives more options.