motioneye-project/motioneyeos #223: How to set VPN PPTP connection

soroushcrazy1, 2016-03-20:
I want to add PPTP connection or L2TP connectio to my raspberry pi motineyeOS. Or installing hamachi VPN on my pi.
my motioneyeOS (raspberry pi 2) has connected to 3G modem that could not get static IP or VPN setting on modem.
I think if We could have VPN connection or logmein hamachi on camera we could have secure and felexible IP camera without static IP for visit camera as far as possible.
please help me how to add pptp connection on last OS ?

CrazyEyesPete, 2016-04-14:
This looks promising and a route I plan to take, but I haven't started using it yet:
https://www.weaved.com/remote-connection-raspberry-pi-linux/

Stunod7, 2016-06-02:
Found this while trying to ask the same question.  motioneyeOS is _not_ a version of linux that you'll be able to do that kind of stuff with.  I wanted to install Hamachi, arrived at this post, then found this: https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-cant-seem-to-find-apt-get-or-any-other-package-manager-what-gives - So I'm probably going to rebuild my box on Raspbian so that I can install Hamachi.  Hope this helps!

bortek, 2017-03-01:
Irrelevant to this project. closing this issue.

closed 2017-03-01 by bortek

andersonid, 2020-06-27:
> Irrelevant to this project. closing this issue.

I really believe that is relevant. The possibility of connect to a VPN and access your cameras remotely is a must have feature.

starbasessd, 2020-06-27:
You can do it using Raspbian/RaspberryPiOS & motion/motioneye install,
allowing the underlying Raspbian handle the VPN connection.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?