motioneye-project/motioneyeos #224: USB File Storage

BrianDurward, 2016-03-22:
Here are the screen shots.
Brian

[image]
[image]

ccrisan, 2016-03-22:
What happens if you navigate to sda1 or sdb1?

BrianDurward, 2016-03-22:
In sda1 I find a current (dated) directory.
Nothing in sdb1.
[image]

I should mention that I used gparted to prepare the USB dive and selected ext3 file types.

ccrisan, 2016-03-22:
So what seems to be the problem?

BrianDurward, 2016-03-22:
I was looking in home/ftp/storage where sda1 and sdb1 are both empty.
[image]

So all is clear - I navigate to /data/media/sda1/myfolder

Many thanks for your quick help.

Brian

ccrisan, 2016-03-22:
Alright :)

closed 2016-03-22 by ccrisan